llvm/circt #1207: [FIRRTL] LowerTypes takes too long on Aliaksei's test3 machine generated example

lattner, 2021-06-05:
I'm seeing this on my laptop:

```
===-------------------------------------------------------------------------===
                         ... Execution time report ...
===-------------------------------------------------------------------------===
  Total Execution Time: 82.5711 seconds

  ----User Time----  ----Wall Time----  ----Name----
    4.3971 (  4.3%)    4.3971 (  5.3%)  FIR Parser
   79.7225 ( 78.6%)   68.2242 ( 82.6%)  'firrtl.circuit' Pipeline
   59.9588 ( 59.1%)   59.9588 ( 72.6%)    LowerFIRRTLTypes
   12.8427 ( 12.7%)    6.0714 (  7.4%)    'firrtl.module' Pipeline
    2.9459 (  2.9%)    1.4290 (  1.7%)      ExpandWhens
    3.5450 (  3.5%)    1.7376 (  2.1%)      CSE
    0.0000 (  0.0%)    0.0000 (  0.0%)        (A) DominanceInfo
    6.3517 (  6.3%)    2.9048 (  3.5%)      SimpleCanonicalizer
    0.9890 (  1.0%)    0.9890 (  1.2%)    IMConstProp
    0.3303 (  0.3%)    0.3303 (  0.4%)    BlackBoxReader
    0.3637 (  0.4%)    0.1542 (  0.2%)    'firrtl.module' Pipeline
    0.3636 (  0.4%)    0.1542 (  0.2%)      CheckWidths
    1.4981 (  1.5%)    1.4981 (  1.8%)  LowerFIRRTLToHW
    1.1365 (  1.1%)    1.1365 (  1.4%)  HWMemSimImpl
    6.0494 (  6.0%)    1.5548 (  1.9%)  'hw.module' Pipeline
    1.2394 (  1.2%)    0.4204 (  0.5%)    HWCleanup
    2.1116 (  2.1%)    0.6226 (  0.8%)    CSE
    0.2884 (  0.3%)    0.0200 (  0.0%)      (A) DominanceInfo
    1.6794 (  1.7%)    0.5579 (  0.7%)    SimpleCanonicalizer
    1.0998 (  1.1%)    1.0998 (  1.3%)  HWLegalizeNames
    1.8463 (  1.8%)    0.4772 (  0.6%)  'hw.module' Pipeline
    1.3443 (  1.3%)    0.4473 (  0.5%)    PrettifyVerilog
    1.8983 (  1.9%)    1.8983 (  2.3%)  Output
    0.0066 (  0.0%)    0.0066 (  0.0%)  Rest
  101.3771 (100.0%)   82.5711 (100.0%)  Total
```

lattner, 2021-06-05:
Suggestions: 
1) We should make this pass iterative instead of "all at once", lowering each level of the types as it goes.
2) `flattenType` isn't cached, and is N^2 in the work it does.  Once this is iterative, it will make it much easier to cache.
3) Because this pass is "all at once", the set/getBundleLowering are taking identifiers for the entire name path to a deeply nested field.  This is inefficient in several ways, not least of which that the identifiers are `mlir::Identifier`'s which are uniqued forever in MLIR and therefore effectively leaked (because there is no reuse).  We're also creating N^2 of them because of the afore mentioned issue.

In short, we should go with an iterative approach which lowers an operations, then rewrites all its users sparsely.  This will greatly improve the efficiency and scalability of the compiler, and will make the code simpler as well.

prithayan, 2021-06-07:
Thanks for the suggestions Chris, working on it.

youngar, 2021-06-07:
I opened PR #1211 to switch from String identifiers to FieldRefs, which speeds us up from 59s to 51s.  It also opens the door to avoid creating all those suffix strings in `flattenType()`.  I think in the future it would make sense to turn `flattenType()` into a visitor pattern, to avoid building up an array of all the leaf elements.

Profiling is showing a lot of time spent lowering SubfieldOps, and a decent amount of time spent lowering MemoryOps.

I put a printf in `setBundleLowering()`, and it /looks/ like the MemOp lowering is incorrectly setting the bundle lowering for `en`, `clk`, `addr` for each lowering of the data field., which could be fixed.
```swift
// void TypeLoweringVisitor::setBundleLowering(FieldRef fieldRef, Value newValue) {
//   llvm::errs() << "setBundleLowering: " << getFieldName(fieldRef) << "\n";
setBundleLowering: bar.r.addr
setBundleLowering: bar.r.en
setBundleLowering: bar.r.clk
setBundleLowering: bar.r.data[0]
setBundleLowering: bar.w.addr
setBundleLowering: bar.w.en
setBundleLowering: bar.w.clk
setBundleLowering: bar.w.data[0]
setBundleLowering: bar.w.mask[0]
setBundleLowering: bar.r.addr
setBundleLowering: bar.r.en
setBundleLowering: bar.r.clk
setBundleLowering: bar.r.data[1]
setBundleLowering: bar.w.addr
setBundleLowering: bar.w.en
setBundleLowering: bar.w.clk
setBundleLowering: bar.w.data[1]
setBundleLowering: bar.w.mask[1]
```