llvm/circt #1209: [FIRRTL] Rename `BlackBoxReader.cpp` or `BlackboxMemory.cpp`

lattner, 2021-06-06:
We should have a consistent casing strategy here, who wins?

youngar, 2021-06-08:
I think "Black Box" makes more sense, and is consistent with annotation names.

lattner, 2021-06-08:
That's a good argument

seldridge, 2021-06-08:
Chisel and FIRRTL have always used `BlackBox` or `blackBox` (implying it's two words). Wikipedia seems to agree: https://en.wikipedia.org/wiki/Black_box. However, adjectives or nouns derived from this concept go different ways, e.g., [`black-box testing`](https://en.wikipedia.org/wiki/Black-box_testing) or [`blackboxing`](https://en.wikipedia.org/wiki/Blackboxing).

I'm good with `BlackBox`, but am fine with either.

youngar, 2021-06-08:
For reference, the pass names are "firrtl-blackbox-reader" and "firrtl-blackbox-memory"

lattner, 2021-06-08:
It sounds like we should go with `BlackBox` or `blackBox` in camel case situations and `blackbox` in all lower case scenarios.  That's simple and consistent.  Who wants to do the honors?

fabianschuiki, 2021-06-09:
:tada: this has been bugging me for a while, thanks!

youngar, 2021-06-09:
closed by #1227

closed 2021-06-09 by youngar