llvm/circt #1215: [FIRRTL] (shrd, and) mismatch

drom, 2021-06-08:
The following FIRRTL program

```python
circuit top_mod :
  module top_mod :
    input inp_e: SInt<58>
    output tmp116: UInt<58>
    wire tmp109: SInt<58>
    tmp109 <= dshr(inp_e, tail(UInt<21>("h1935b6"), 11))
    tmp116 <= and(asSInt(UInt<9>("o246")), tmp109)
```

Compiled with `firtool` produces this Verilog:

```verilog
module top_mod(
  input  [57:0] inp_e,
  output [57:0] tmp116);

  assign tmp116 = $signed(inp_e) >>> 58'sh1B6 & 58'hA6;	// a_top_mod.fir:2:3, :6:15, :7:15
endmodule
```

Compiled with `firrtl-1.5-SNAPSHOT` produces this Verilog:

```verilog
module top_mod(
  input  [57:0] inp_e,
  output [57:0] tmp116
);
  wire  _GEN_1 = inp_e[57];
  wire [57:0] tmp109 = {58{_GEN_1}};
  assign tmp116 = 58'sha6 & $signed(tmp109);
endmodule
```

`Yosys 0.9+3911 (git sha1 4b31223e, clang 11.0.1 -fPIC -Os)` reports formal mismatch:

```
ERROR: Found 4 unproven $equiv cells in 'equiv_status -assert'.
```

drom, 2021-06-08:
Related issue:
The following FIRRTL program

```python
circuit top_mod :
  module top_mod :
    input tmp15: SInt<67>
    input tmp37: UInt<1>
    output _tmp66: UInt<67>
    wire tmp23: SInt<67>
    tmp23 <= dshr(tmp15, head(asUInt(SInt<14>(2715)), 5))
    _tmp66 <= xor(tmp37, asUInt(tmp23))
```

Compiled with `firtool` produces this Verilog:

```verilog
module top_mod(
  input  [66:0] tmp15,
  input         tmp37,
  output [66:0] _tmp66);

  assign _tmp66 = {66'h0, tmp37} ^ $signed(tmp15) >>> 67'sh5;	// a_top_mod.fir:2:3, :7:14, :8:15
endmodule
```

Compiled with `firrtl-1.5-SNAPSHOT` produces this Verilog:

```verilog
module top_mod(
  input  [66:0] tmp15,
  input         tmp37,
  output [66:0] _tmp66
);
  wire [61:0] _GEN_0 = tmp15[66:5];
  wire [66:0] _GEN_1 = {{66'd0}, tmp37};
  wire [66:0] _GEN_2 = {{5{_GEN_0[61]}},_GEN_0};
  assign _tmp66 = _GEN_1 ^ _GEN_2;
endmodule
```

`Yosys 0.9+3911 (git sha1 4b31223e, clang 11.0.1 -fPIC -Os)` reports formal mismatch:

```
ERROR: Found 5 unproven $equiv cells in 'equiv_status -assert'.
```

darthscsi, 2021-06-09:
@drom tried adding parenthesis to check for precedence problems, and this doesn't appear to be a precedence problem.

drom, 2021-06-09:
yep. I did the same. It is not.

fabianschuiki, 2021-06-11:
This could be related to some issue in Yosys, or an edge case of SV I don't understand. Usually the type and sign of an expression is determined entirely by the operand types and signs, whereas the bit width is determined by the operand and the assignment-like context (i.e. the left-hand side of the assignment). Interestingly, the circuits match if I replace the MFC with:
```verilog
  wire [66:0] foo = $signed(tmp15) >>> 67'sh5;
  assign _tmp66 = {66'h0, tmp37} ^ foo; // a_top_mod.fir:2:3, :7:14, :8:15
```
or also with
```verilog
  assign _tmp66 = {66'h0, tmp37} ^ $unsigned($signed(tmp15) >>> 67'sh5); // a_top_mod.fir:2:3, :7:14, :8:15
```
or also with
```verilog
  assign _tmp66 = {66'h0, tmp37} ^ $signed($signed(tmp15) >>> 67'sh5); // a_top_mod.fir:2:3, :7:14, :8:15
```
But interestingly, just parenthesizing the shift (which is unnecessary according to precedence rules, as @drom also pointed out), equivalence checking fails:
```verilog
  assign _tmp66 = {66'h0, tmp37} ^ ($signed(tmp15) >>> 67'sh5); // a_top_mod.fir:2:3, :7:14, :8:15
```

Pretty strange.

darthscsi, 2021-06-11:
For some other problems like this, I've tried reducing the bitwidths as
much as possible, then building a test bench to compare the outputs
exhaustively in simulation.

lattner, 2021-06-13:
I'm pretty sure this is a yosis problem not handling extremely large shift amounts.

lattner, 2021-06-13:
With f0281f62 we now produce:

```
module top_mod(
  input  [57:0] inp_e,
  output [57:0] tmp116);

  assign tmp116 = {58{inp_e[57]}} & 58'hA6;	// x.fir:2:3, :6:15, :7:15
endmodule
```

I think this should compare the same with yosis now.

closed 2021-06-13 by lattner