llvm/circt #1220: [FIRRTL] quality of emitted verilog

drom, 2021-06-08:
The following FIRRTL program

```python
; seed: 41635
circuit top_mod :
  module top_mod :
    input clock: Clock
    input reset: UInt<1>
    input arst: AsyncReset
    input inp_a: {inp_b: {inp_j: {inp_l: {inp_n: {inp_cb: UInt<36>}}}}, inp_e: {inp_k: SInt<57>}}
    input inp_c: {inp_d: {inp_f: {inp_o: SInt<8>, inp_p: SInt<65>, inp_ab: SInt<26>}}[1][5], inp_g: {inp_m: {inp_db: SInt<56>}, inp_bb: SInt<13>}, inp_i: UInt<7>}
    input inp_h: SInt<64>[1]
    output tmp21: UInt<26>
    output _tmp22: UInt<26>
    output tmp23: SInt<56>
    output fsm_24_next: UInt<1>
    output tmp26: SInt<16>
    output tmp27: UInt<1>
    output tmp31: UInt<4>
    output tmp32: SInt<21>
    output tmp35: SInt<57>
    reg fsm_24_state: UInt<1>, clock with: (reset => (arst, UInt<1>(0)))
    tmp21 <= or(inp_c.inp_d[4][0].inp_f.inp_o, inp_c.inp_d[4][0].inp_f.inp_ab)
    _tmp22 <= or(inp_c.inp_d[4][0].inp_f.inp_ab, SInt<14>("o5464"))
    tmp23 <= pad(inp_c.inp_g.inp_m.inp_db, 40)
    when eq(fsm_24_state, UInt<1>(0)) :
      when xorr(inp_c.inp_g.inp_bb) :
        fsm_24_next <= UInt<1>(1)
      else :
        fsm_24_next <= fsm_24_state
    else when eq(fsm_24_state, UInt<1>(1)) :
      when xorr(UInt<21>("b10110100001101100100")) :
        fsm_24_next <= UInt<1>(0)
      else :
        fsm_24_next <= fsm_24_state
    else :
      fsm_24_next <= fsm_24_state
    fsm_24_state <= fsm_24_next
    tmp26 <= SInt<16>("o33007")
    tmp27 <= andr(inp_c.inp_d[4][0].inp_f.inp_p)
    tmp31 <= tail(inp_c.inp_d[4][0].inp_f.inp_o, 4)
    tmp32 <= mul(SInt<13>("h-44a"), SInt<8>(110))
    skip
    tmp35 <= add(inp_c.inp_g.inp_m.inp_db, SInt<25>(-10246229))
    skip
```

Compiled with `firtool` produces this Verilog:

```verilog
// Standard header to adapt well known macros to our needs.
`ifdef RANDOMIZE_REG_INIT
  `define RANDOMIZE
`endif
`ifndef RANDOM
  `define RANDOM $random
`endif

// Users can define INIT_RANDOM as general code that gets injected into the
// initializer block for modules with registers.
`ifndef INIT_RANDOM
  `define INIT_RANDOM
`endif

// If using random initialization, you can also define RANDOMIZE_DELAY to
// customize the delay used, otherwise 0.002 is used.
`ifndef RANDOMIZE_DELAY
  `define RANDOMIZE_DELAY 0.002
`endif

// Define INIT_RANDOM_PROLOG_ for use in our modules below.
`ifdef RANDOMIZE
  `ifdef VERILATOR
    `define INIT_RANDOM_PROLOG_ `INIT_RANDOM
  `else
    `define INIT_RANDOM_PROLOG_ `INIT_RANDOM #`RANDOMIZE_DELAY begin end
  `endif
`else
  `define INIT_RANDOM_PROLOG_
`endif

module top_mod(
  input         clock, reset, arst,
  input  [35:0] inp_a_inp_b_inp_j_inp_l_inp_n_inp_cb,
  input  [56:0] inp_a_inp_e_inp_k,
  input  [7:0]  inp_c_inp_d_0_0_inp_f_inp_o,
  input  [64:0] inp_c_inp_d_0_0_inp_f_inp_p,
  input  [25:0] inp_c_inp_d_0_0_inp_f_inp_ab,
  input  [7:0]  inp_c_inp_d_1_0_inp_f_inp_o,
  input  [64:0] inp_c_inp_d_1_0_inp_f_inp_p,
  input  [25:0] inp_c_inp_d_1_0_inp_f_inp_ab,
  input  [7:0]  inp_c_inp_d_2_0_inp_f_inp_o,
  input  [64:0] inp_c_inp_d_2_0_inp_f_inp_p,
  input  [25:0] inp_c_inp_d_2_0_inp_f_inp_ab,
  input  [7:0]  inp_c_inp_d_3_0_inp_f_inp_o,
  input  [64:0] inp_c_inp_d_3_0_inp_f_inp_p,
  input  [25:0] inp_c_inp_d_3_0_inp_f_inp_ab,
  input  [7:0]  inp_c_inp_d_4_0_inp_f_inp_o,
  input  [64:0] inp_c_inp_d_4_0_inp_f_inp_p,
  input  [25:0] inp_c_inp_d_4_0_inp_f_inp_ab,
  input  [55:0] inp_c_inp_g_inp_m_inp_db,
  input  [12:0] inp_c_inp_g_inp_bb,
  input  [6:0]  inp_c_inp_i,
  input  [63:0] inp_h_0,
  output [25:0] tmp21, _tmp22,
  output [55:0] tmp23,
  output        fsm_24_next,
  output [15:0] tmp26,
  output        tmp27,
  output [3:0]  tmp31,
  output [20:0] tmp32,
  output [56:0] tmp35);

  wire _T;	// a_top_mod.fir:34:19
  reg  fsm_24_state;	// a_top_mod.fir:19:5

  localparam _T_0 = 1'h1;	// a_top_mod.fir:25:24
  localparam _T_1 = 1'h0;	// a_top_mod.fir:19:61
  always @(posedge clock or posedge arst) begin	// a_top_mod.fir:19:5
    if (arst)	// a_top_mod.fir:19:5
      fsm_24_state <= _T_1;	// a_top_mod.fir:19:5
    else	// a_top_mod.fir:19:5
      fsm_24_state <= _T;	// a_top_mod.fir:34:19, :35:18
  end // always @(posedge or posedge)
  `ifndef SYNTHESIS	// a_top_mod.fir:19:5
    initial begin	// a_top_mod.fir:19:5
      `INIT_RANDOM_PROLOG_	// a_top_mod.fir:19:5
      `ifdef RANDOMIZE_REG_INIT	// a_top_mod.fir:19:5
        if (~arst)	// a_top_mod.fir:19:5
          fsm_24_state = `RANDOM;	// a_top_mod.fir:19:5
      `endif
    end // initial
  `endif
  wire _T_2 = fsm_24_state;	// a_top_mod.fir:23:10
  assign _T = ~_T_2 ? ~_T_2 & ^inp_c_inp_g_inp_bb | _T_2 : ~(~(~_T_2) & _T_2) & _T_2;	// a_top_mod.fir:23:{5,10}, :24:12, :27:21, :34:19
  wire [56:0] _tmp_3 = {inp_c_inp_g_inp_m_inp_db[55], inp_c_inp_g_inp_m_inp_db};	// a_top_mod.fir:41:14
  wire [25:0] _tmp = {{18{inp_c_inp_d_4_0_inp_f_inp_o[7]}}, inp_c_inp_d_4_0_inp_f_inp_o};	// a_top_mod.fir:20:14
  assign tmp21 = _tmp | inp_c_inp_d_4_0_inp_f_inp_ab;	// a_top_mod.fir:3:3, :20:14
  assign _tmp22 = inp_c_inp_d_4_0_inp_f_inp_ab | 26'hB34;	// a_top_mod.fir:3:3, :21:15
  assign tmp23 = inp_c_inp_g_inp_m_inp_db;	// a_top_mod.fir:3:3
  assign fsm_24_next = _T;	// a_top_mod.fir:3:3, :34:19
  assign tmp26 = 16'h3607;	// a_top_mod.fir:3:3, :36:14
  assign tmp27 = &inp_c_inp_d_4_0_inp_f_inp_p;	// a_top_mod.fir:3:3, :37:14
  assign tmp31 = inp_c_inp_d_4_0_inp_f_inp_o[3:0];	// a_top_mod.fir:3:3, :38:14
  assign tmp32 = 21'h1E2834;	// a_top_mod.fir:3:3, :39:14
  assign tmp35 = _tmp_3 + 57'h1FFFFFFFF63A7AB;	// a_top_mod.fir:3:3, :41:14
endmodule
```

Compiled with `firrtl-1.5-SNAPSHOT` produces this Verilog:

```verilog
module top_mod(
  input         clock,
  input         reset,
  input         arst,
  input  [35:0] inp_a_inp_b_inp_j_inp_l_inp_n_inp_cb,
  input  [56:0] inp_a_inp_e_inp_k,
  input  [7:0]  inp_c_inp_d_0_0_inp_f_inp_o,
  input  [64:0] inp_c_inp_d_0_0_inp_f_inp_p,
  input  [25:0] inp_c_inp_d_0_0_inp_f_inp_ab,
  input  [7:0]  inp_c_inp_d_1_0_inp_f_inp_o,
  input  [64:0] inp_c_inp_d_1_0_inp_f_inp_p,
  input  [25:0] inp_c_inp_d_1_0_inp_f_inp_ab,
  input  [7:0]  inp_c_inp_d_2_0_inp_f_inp_o,
  input  [64:0] inp_c_inp_d_2_0_inp_f_inp_p,
  input  [25:0] inp_c_inp_d_2_0_inp_f_inp_ab,
  input  [7:0]  inp_c_inp_d_3_0_inp_f_inp_o,
  input  [64:0] inp_c_inp_d_3_0_inp_f_inp_p,
  input  [25:0] inp_c_inp_d_3_0_inp_f_inp_ab,
  input  [7:0]  inp_c_inp_d_4_0_inp_f_inp_o,
  input  [64:0] inp_c_inp_d_4_0_inp_f_inp_p,
  input  [25:0] inp_c_inp_d_4_0_inp_f_inp_ab,
  input  [55:0] inp_c_inp_g_inp_m_inp_db,
  input  [12:0] inp_c_inp_g_inp_bb,
  input  [6:0]  inp_c_inp_i,
  input  [63:0] inp_h_0,
  output [25:0] tmp21,
  output [25:0] _tmp22,
  output [55:0] tmp23,
  output        fsm_24_next,
  output [15:0] tmp26,
  output        tmp27,
  output [3:0]  tmp31,
  output [20:0] tmp32,
  output [56:0] tmp35
);
`ifdef RANDOMIZE_REG_INIT
  reg [31:0] _RAND_0;
`endif // RANDOMIZE_REG_INIT
  reg  fsm_24_state;
  wire  _GEN_2 = fsm_24_state ? 1'h0 : fsm_24_state; // @[]
  wire [25:0] _GEN_5 = {{18{inp_c_inp_d_4_0_inp_f_inp_o[7]}},inp_c_inp_d_4_0_inp_f_inp_o};
  assign tmp21 = $signed(_GEN_5) | $signed(inp_c_inp_d_4_0_inp_f_inp_ab);
  assign _tmp22 = $signed(inp_c_inp_d_4_0_inp_f_inp_ab) | 26'shb34;
  assign tmp23 = inp_c_inp_g_inp_m_inp_db;
  assign fsm_24_next = ~fsm_24_state ? ^inp_c_inp_g_inp_bb | fsm_24_state : _GEN_2; // @[]
  assign tmp26 = 16'sh3607;
  assign tmp27 = $signed(inp_c_inp_d_4_0_inp_f_inp_p) == -65'sh1;
  assign tmp31 = inp_c_inp_d_4_0_inp_f_inp_o[3:0];
  assign tmp32 = -13'sh44a * 8'sh6e;
  assign tmp35 = $signed(inp_c_inp_g_inp_m_inp_db) - 56'sh9c5855;
  always @(posedge clock or posedge arst) begin
    if (arst) begin
      fsm_24_state <= 1'h0;
    end else begin
      fsm_24_state <= fsm_24_next;
    end
  end
// Register and memory initialization
`ifdef RANDOMIZE_GARBAGE_ASSIGN
`define RANDOMIZE
`endif
`ifdef RANDOMIZE_INVALID_ASSIGN
`define RANDOMIZE
`endif
`ifdef RANDOMIZE_REG_INIT
`define RANDOMIZE
`endif
`ifdef RANDOMIZE_MEM_INIT
`define RANDOMIZE
`endif
`ifndef RANDOM
`define RANDOM $random
`endif
`ifdef RANDOMIZE_MEM_INIT
  integer initvar;
`endif
`ifndef SYNTHESIS
`ifdef FIRRTL_BEFORE_INITIAL
`FIRRTL_BEFORE_INITIAL
`endif
initial begin
  `ifdef RANDOMIZE
    `ifdef INIT_RANDOM
      `INIT_RANDOM
    `endif
    `ifndef VERILATOR
      `ifdef RANDOMIZE_DELAY
        #`RANDOMIZE_DELAY begin end
      `else
        #0.002 begin end
      `endif
    `endif
`ifdef RANDOMIZE_REG_INIT
  _RAND_0 = {1{`RANDOM}};
  fsm_24_state = _RAND_0[0:0];
`endif // RANDOMIZE_REG_INIT
  if (arst) begin
    fsm_24_state = 1'h0;
  end
  `endif // RANDOMIZE
end // initial
`ifdef FIRRTL_AFTER_INITIAL
`FIRRTL_AFTER_INITIAL
`endif
`endif // SYNTHESIS
endmodule
```

drom, 2021-06-08:
Small constants got confusing, counteractive names:

```verilog
  localparam _T_0 = 1'h1;	// a_top_mod.fir:25:24
  localparam _T_1 = 1'h0;	// a_top_mod.fir:19:61
```

drom, 2021-06-08:
Multiple unnecessary renamings of the signal:

```verilog
  wire _T_2 = fsm_24_state;	// a_top_mod.fir:23:10
  ...
  assign _T = ~_T_2 ? ~_T_2 & ^inp_c_inp_g_inp_bb | _T_2 : ~(~(~_T_2) & _T_2) & _T_2;	// a_top_mod.fir:23:{5,10}, :24:12, :27:21, :34:19
  ...
  assign fsm_24_next = _T;	// a_top_mod.fir:3:3, :34:19
```

while SFC is doing better job of keeping the names:

```verilog
  wire  _GEN_2 = fsm_24_state ? 1'h0 : fsm_24_state; // @[]
  assign fsm_24_next = ~fsm_24_state ? ^inp_c_inp_g_inp_bb | fsm_24_state : _GEN_2; // @[]
```

drom, 2021-06-08:
Expressions that can be further reduced:

```verilog
assign _T = ~_T_2 ? ~_T_2 & ^inp_c_inp_g_inp_bb | _T_2 : ~(~(~_T_2) & _T_2) & _T_2;
```

drom, 2021-06-08:
How it was:

```verilog
  wire _T_2 = fsm_24_state;	// a_top_mod.fir:23:10
  ...
  assign _T = ~_T_2 ? ~_T_2 & ^inp_c_inp_g_inp_bb | _T_2 : ~(~(~_T_2) & _T_2) & _T_2;	// a_top_mod.fir:23:{5,10}, :24:12, :27:21, :34:19
  ...
  assign fsm_24_next = _T;	// a_top_mod.fir:3:3, :34:19
```

How it is after https://github.com/llvm/circt/commit/4e524d0e8f6426b0a9f1d87e926062616dc17d7d

```verilog
  wire _T_2 = fsm_24_state;	// a_top_mod.fir:23:10
...
  assign _T = _T_2 ? ~(~(~_T_2) & _T_2) & _T_2 : ~_T_2 & ^inp_c_inp_g_inp_bb | _T_2;	// a_top_mod.fir:23:{5,10}, :24:12, :27:21, :34:19
...
  assign fsm_24_next = _T;
```

drom, 2021-06-08:
How it was:

```verilog
  wire _T_2 = fsm_24_state;	// a_top_mod.fir:23:10
...
  assign _T = _T_2 ? ~(~(~_T_2) & _T_2) & _T_2 : ~_T_2 & ^inp_c_inp_g_inp_bb | _T_2;	// a_top_mod.fir:23:{5,10}, :24:12, :27:21, :34:19
...
  assign fsm_24_next = _T;
```

How it is after https://github.com/llvm/circt/commit/53e986eb5d6a85141ff49c9abb8b4225efc0113b

```verilog
  wire _T_2 = fsm_24_state;	// a_top_mod.fir:23:10
  ...
  assign _T = _T_2 ? ~_T_2 & _T_2 : ~_T_2 & ^inp_c_inp_g_inp_bb | _T_2;	// a_top_mod.fir:23:10, :24:12, :27:21, :34:19
  ...
  assign fsm_24_next = _T;	// a_top_mod.fir:3:3, :34:19
```

drom, 2021-06-08:
How it was:

```verilog
  assign _T = _T_2 ? ~_T_2 & _T_2 : ~_T_2 & ^inp_c_inp_g_inp_bb | _T_2;	// a_top_mod.fir:23:10, :24:12, :27:21, :34:19
```

How it is after https://github.com/llvm/circt/commit/bdea3d5e17f2938c9dbc802cda60a8a96a60893f

```verilog
  assign _T = ~_T_2 & (~_T_2 & ^inp_c_inp_g_inp_bb | _T_2);	// a_top_mod.fir:23:10, :24:12, :27:21, :34:19
```

lattner, 2021-06-08:
[This](https://github.com/llvm/circt/issues/1223) is why the constants are getting emitted into their own temporaries.

I agree that we're missing some folds/canonicalizations here with the ~'s.  These should be done at the comb level.

youngar, 2021-06-22:
Sinking of constants have been implemented, this is the current output:
```verilog
  wire _T;	// ./dromtest.fir:34:19
  reg  fsm_24_state;	// ./dromtest.fir:19:5

  always_ff @(posedge clock or posedge arst) begin	// ./dromtest.fir:19:5
    if (arst)	// ./dromtest.fir:19:5
      fsm_24_state <= 1'h0;	// ./dromtest.fir:19:{5,61}
    else	// ./dromtest.fir:19:5
      fsm_24_state <= _T;	// ./dromtest.fir:34:19, :35:18
  end // always_ff @(posedge or posedge)
  `ifndef SYNTHESIS	// ./dromtest.fir:19:5
    initial begin	// ./dromtest.fir:19:5
      `INIT_RANDOM_PROLOG_	// ./dromtest.fir:19:5
      `ifdef RANDOMIZE_REG_INIT	// ./dromtest.fir:19:5
        if (~arst)	// ./dromtest.fir:19:5
          fsm_24_state = `RANDOM;	// ./dromtest.fir:19:5
      `endif
    end // initial
  `endif
  wire _T_0 = fsm_24_state;	// ./dromtest.fir:23:10
  assign _T = ~_T_0 & (~_T_0 & ^inp_c_inp_g_inp_bb | _T_0);	// ./dromtest.fir:23:10, :24:12, :27:21, :34:19
  wire [56:0] _tmp_1 = {inp_c_inp_g_inp_m_inp_db[55], inp_c_inp_g_inp_m_inp_db};	// ./dromtest.fir:41:14
  wire [25:0] _tmp = {{18{inp_c_inp_d_4_0_inp_f_inp_o[7]}}, inp_c_inp_d_4_0_inp_f_inp_o};	// ./dromtest.fir:20:14
  assign tmp21 = _tmp | inp_c_inp_d_4_0_inp_f_inp_ab;	// ./dromtest.fir:3:10, :20:14
  assign _tmp22 = inp_c_inp_d_4_0_inp_f_inp_ab | 26'hB34;	// ./dromtest.fir:3:10, :21:15
  assign tmp23 = inp_c_inp_g_inp_m_inp_db;	// ./dromtest.fir:3:10
  assign fsm_24_next = _T;	// ./dromtest.fir:3:10, :34:19
  assign tmp26 = 16'h3607;	// ./dromtest.fir:3:10, :36:14
  assign tmp27 = &inp_c_inp_d_4_0_inp_f_inp_p;	// ./dromtest.fir:3:10, :37:14
  assign tmp31 = inp_c_inp_d_4_0_inp_f_inp_o[3:0];	// ./dromtest.fir:3:10, :38:14
  assign tmp32 = 21'h1E2834;	// ./dromtest.fir:3:10, :39:14
  assign tmp35 = _tmp_1 + 57'h1FFFFFFFF63A7AB;	// ./dromtest.fir:3:10, :41:14
  ```

lattner, 2021-08-29:
This is all cleaned up now, including eliminating the _T_0 temp above:

```
 reg fsm_24_state;	// x.fir:18:5

  `ifndef SYNTHESIS	// x.fir:18:5
    initial begin	// x.fir:18:5
      `INIT_RANDOM_PROLOG_	// x.fir:18:5
      `ifdef RANDOMIZE_REG_INIT	// x.fir:18:5
        if (~arst) begin	// x.fir:18:5
          automatic logic [31:0] _T_0 = `RANDOM;	// x.fir:18:5

          fsm_24_state = _T_0[0];	// x.fir:18:5
        end
      `endif
    end // initial
  `endif
  wire _T = fsm_24_state ? 1'h0 : ^inp_c_inp_g_inp_bb | fsm_24_state;	// x.fir:18:61, :22:10, :23:12, :26:21, :33:19
  always @(posedge clock or posedge arst) begin	// x.fir:18:5
    if (arst)	// x.fir:18:5
      fsm_24_state <= 1'h0;	// x.fir:18:{5,61}
    else	// x.fir:18:5
      fsm_24_state <= _T;	// x.fir:34:18
  end // always @(posedge or posedge)
  wire [25:0] _tmp = {{18{inp_c_inp_d_4_0_inp_f_inp_o[7]}}, inp_c_inp_d_4_0_inp_f_inp_o};	// x.fir:19:14
  wire [56:0] _tmp_1 = {inp_c_inp_g_inp_m_inp_db[55], inp_c_inp_g_inp_m_inp_db};	// x.fir:40:14
  assign tmp21 = _tmp | inp_c_inp_d_4_0_inp_f_inp_ab;	// x.fir:2:10, :19:14
  assign _tmp22 = inp_c_inp_d_4_0_inp_f_inp_ab | 26'hB34;	// x.fir:2:10, :20:15
  assign tmp23 = inp_c_inp_g_inp_m_inp_db;	// x.fir:2:10
  assign fsm_24_next = _T;	// x.fir:2:10
  assign tmp26 = 16'h3607;	// x.fir:2:10, :35:14
  assign tmp27 = &inp_c_inp_d_4_0_inp_f_inp_p;	// x.fir:2:10, :36:14
  assign tmp31 = inp_c_inp_d_4_0_inp_f_inp_o[3:0];	// x.fir:2:10, :37:14
  assign tmp32 = 21'h1E2834;	// x.fir:2:10, :38:14
  assign tmp35 = _tmp_1 - 57'h9C5855;	// x.fir:2:10, :40:14
endmodule
```

closed 2021-08-29 by lattner