llvm/circt #1224: [firtool] crashing while parsing annotations

youngar, 2021-06-09:
An empty target string looks like its crashing firtool: `firtool ./test.fir --annotation-file=test.json`.
```json
// test.json:
[
    {
        "target":""
    }
]
```
```
  * frame #4: 0x00000001000a93e5 firtool`llvm::StringRef::operator[](this=<unavailable>, Index=<unavailable>) const at StringRef.h:255:7 [opt]
    frame #5: 0x0000000100461779 firtool`canonicalizeTarget(target=(Data = "", Length = 0)) at FIRAnnotations.cpp:69:7 [opt]
    frame #6: 0x0000000100457f4a firtool`circt::firrtl::fromJSON(this=<unavailable>, object=<unavailable>, p=Path @ 0x00007ffeefbfc700)::$_0::operator()(llvm::json::Object*, llvm::json::Path) const at FIRAnnotations.cpp:163:19 [opt]
```

seldridge, 2021-06-09:
Good find. This will now produce:

```
Tmp.fir:1:14: error: Invalid/unsupported annotation format
circuit Foo: %[[{"target":""}]]
             ^
Tmp.fir:1:14: note: See inline comments for problem area in JSON:
[
  {
    "target": /* error: invalid target string */ ""
  }
]
```

closed 2021-06-09 by seldridge

youngar, 2021-06-09:
Awesome! thanks for the quick fix! unblocks a problem i was having :)

commit 62113838 (2021-06-09): [FIRRTL] Fix crash on invalid Annotation target

Add defensive code to check the validity of an Annotation Target.  If
the annotation is invalid (e.g., an empty string) or the wrong
type (e.g., null), then return a JSON error.  Add tests of both of these
behaviors.

Fixes #1224.

h/t @youngar