llvm/circt #1235: [FIRRTL] BlackBoxReader: something going on with resource paths

youngar, 2021-06-10:
When there is an annotation like this:
```json
{"class": "firrtl.transforms.BlackBoxResourceAnno", "resourceId": "/example.v"}
```
The pass succeeds when using a relative path for the input fir file, e.g. `./test.fir`, but fails after removing the prefixed `./`, e.g. `test.fir`

This is the error message:
```sh
test.fir:4:13: error: cannot open input file '/example.v': No such file or directory
  extmodule Example :
            ^
test.fir:4:13: note: see current operation: "firrtl.extmodule"() ( {
}) {annotations = [{class = "firrtl.transforms.BlackBoxResourceAnno", resourceId = "/example.v"}], arg_attrs = [], portDirections = false, portNames = [], sym_name = "Example", type = () -> ()} : () -> ()
```

Some extra steps:
```scala
; test.fir
circuit Foo: %[[{"class":"firrtl.transforms.BlackBoxResourceAnno","resourceId":"/example.v", "target":"~Foo|Example"}]]
  module Foo:
    skip
  extmodule Example :
```

```sh
# create empty v
touch ./example.v
# error
./bin/firtool test.fir
# success
./bin/firtool ./test.fir
```

fabianschuiki, 2021-06-11:
Huh interesting. There was some special casing for resource annos in the reader that always treated resources as relative paths. This might have gotten mixed up somewhere :thinking:.

fabianschuiki, 2021-06-11:
Ah I see the problem. The resource search path is empty by default. That means it will just use `""` as the prefix to search fo resources, which then makes the `/test.fir` look absolute. There's some code in firtool that tries to infer the blackbox search path from the input file, which it infers as `""` and `"."` in your two `firtool` invocations. I'll whip up a fix.

commit 677fc64a (2021-06-11): [FIRRTL] Fix blackbox resource path no treated relative on empty prefix

* Fix an issue where the blackbox reader pass would not treat a blackbox
  resource path as relative when the resource prefix was the empty path.
  This could occur for example when running `firtool test.fir`, where
  the directory prefix of the input file `test.fir` was an empty string.
  In that case, a resource path `/foo.v` would come out as an absolute
  path `/foo.v`, instead of the intended `foo.v`. Running firtool with
  `firtool ./test.fir` would mitigate the issue, since the directory
  prefix would be `.`, conc [... cut]
closed 2021-06-11 by fabianschuiki