llvm/circt #1236: [FIRRTL][IMCP] IMCostProp leaving invalid modules around

youngar, 2021-06-10:
IMConstProp is stripping out the bodies of modules which are never instantiated:

```scala
circuit Top :
  module Test2 :
    output out : UInt<16>
    out <= UInt<16>("h222")
  module Top :
```

```mlir
// firtool --imconstprop=true ./test.fir
module  {
  firrtl.circuit "Top"  {
    firrtl.module @Test2(out %out: !firrtl.uint<16>) {
      %c546_ui16 = firrtl.constant 546 : !firrtl.uint<16>
      firrtl.connect %out, %c546_ui16 : !firrtl.uint<16>, !firrtl.uint<16>
    }
    firrtl.module @Top() {
    }
  }
}
```

The questions are:

1. Is it illegal to have modules with nothing driving the output?
    It would not pass ExpandWhens initialization checking so I am thinking yes.
2. Should IMCP delete the entire module instead of leaving it around?
    Probably.
3. If a Node inside a module is marked DoNotTouch, can we delete the entire module? (even if it is never instantiated)
    Will check with the CAAT team

youngar, 2021-06-10:
From Schuyler:

---

Example for behavior of DontTouchAnnotation on components in modules that are not instantiated: "The component don't touched is preserved, but anything else may be optimized away." E.g.:
```scala
circuit Foo :
  module Bar :
    input deleteMe: UInt<1>
    wire a: UInt<1>
    a is invalid
  module Foo :
    skip
```
With annotation file:
```json
[
    {"class":"firrtl.transforms.DontTouchAnnotation",
     "target":"~Foo|Bar>a"}
]
```
Produces:
```verilog
module Bar(
);
  wire  a = 1'h0;
endmodule
module Foo(
);
endmodule
```

lattner, 2021-06-14:
Good point, this shouldn't be the domain of imconstprop.  Fixed in b9ac3646, thanks!

closed 2021-06-14 by lattner