llvm/circt #1243: Run CI test through valgrind?

fabianschuiki, 2021-06-11:
I just ran into an out-of-bounds access I accidentally introduced in #1178, which was easy to catch through Valgrind, but wasn't obvious from the CI run (and I probably got lucky that it triggered a crash). Would it make sense to also run the integration tests through valgrind in CI, to systematically check for leaks?

teqdruid, 2021-06-11:
We should definitely run the unittests with valgrind on a nightly basis (at least). Some of the integration tests are pretty long, so we might think twice about that. Also, python and valgrind have a... complicated relationship so we shouldn't run those with valgrind.

fabianschuiki, 2021-06-11:
Yeah I think the unit tests would already be super helpful!

mikeurbach, 2021-06-11:
Since we're already building some of the unit tests with clang, could we potentially enable [AddressSanitizer](https://clang.llvm.org/docs/AddressSanitizer.html)? I think that would check for this specific case, and would be one less dependency to have in our test container.

youngar, 2021-06-11:
I think it would be great to have ASAN or valgrind in the PR builds. I have been building with ASAN locally, and the build times for LLVM are pretty long but its not bad for incremental builds. Tests take longer but its still short. 

There is an open issue for LLHD that is semi-blocking this: https://github.com/llvm/circt/issues/1071

fabianschuiki, 2021-06-14:
> There is an open issue for LLHD that is semi-blocking this: #1071

Ah good point. I'll try to fix that asap.

dtzSiFive, 2022-10-11:
At least for the lit-based tests, lit has support for running them under valgrind (`--vg`) which we could pretty easily add in our CI runs if we wanted (`LLVM_LIT_ARGS`, or run 'lit' ourselves with that flag pointed at the desired test directory / config).

Building everything with asan + ubsan is my preference (and how I usually test things) and would be great to add (!) but enabling valgrind checking seems like an easy place to start for now.