llvm/circt #1255: [FIRRTL] infer width (tail, constant) (core dump)

drom, 2021-06-12:
The following FIRRTL program:

```python
circuit top_mod :
  module top_mod :
    wire tmp74: UInt
    tmp74 <= tail(UInt("b11101001111100"), 15)
```

Compiled with `firtool a_top_mod.fir --lower-to-hw --expand-whens --infer-widths --lowering-options=noAlwaysFF --verilog -o=a_top_mod_new.v` fails with the following core dump:

```
firtool: ../lib/Dialect/FIRRTL/Transforms/InferWidths.cpp:1284: bool {anonymous}::InferenceTypeUpdate::updateValue(mlir::Value): Assertion `solution >= 0' failed.
PLEASE submit a bug report to https://bugs.llvm.org/ and include the crash backtrace.
Stack dump:
0.      Program arguments: ../../llvm/circt/build/bin/firtool a_top_mod.fir --lower-to-hw --imconstprop --infer-widths --lowering-options=noAlwaysFF --mlir-timing --verilog -o=a_top_mod_new.v
1.       #0 0x0000000000485b6d llvm::sys::PrintStackTrace(llvm::raw_ostream&, int) (../../llvm/circt/build/bin/firtool+0x485b6d)
 #1 0x00000000004839a4 llvm::sys::RunSignalHandlers() (../../llvm/circt/build/bin/firtool+0x4839a4)
 #2 0x0000000000483b03 SignalHandler(int) (../../llvm/circt/build/bin/firtool+0x483b03)
 #3 0x000000000488c970 __restore_rt (/lib64/libpthread.so.0+0x13970)
 #4 0x0000000004cad322 raise (/lib64/libc.so.6+0x3d322)
 #5 0x0000000004c96864 abort (/lib64/libc.so.6+0x26864)
 #6 0x0000000004c96749 _nl_load_domain.cold (/lib64/libc.so.6+0x26749)
 #7 0x0000000004ca59d6 (/lib64/libc.so.6+0x359d6)
 #8 0x00000000005b8c48 (anonymous namespace)::InferenceTypeUpdate::updateValue(mlir::Value) /home/drom/work/github/llvm/circt/build/../lib/Dialect/FIRRTL/Transforms/InferWidths.cpp:1287:28
 #9 0x00000000005b7ffb (anonymous namespace)::InferenceTypeUpdate::updateOperation(mlir::Operation*) /home/drom/work/github/llvm/circt/build/../lib/Dialect/FIRRTL/Transforms/InferWidths.cpp:1167:16
#10 0x00000000005b7ed0 (anonymous namespace)::InferenceTypeUpdate::update(circt::firrtl::CircuitOp)::'lambda'(mlir::Operation*)::operator()(mlir::Operation*) const /home/drom/work/github/llvm/circt/build/../lib/Dialect/FIRRTL/Transforms/InferWidths.cpp:1157:20
#11 0x00000000005c91d3 mlir::WalkResult llvm::function_ref<mlir::WalkResult (mlir::Operation*)>::callback_fn<(anonymous namespace)::InferenceTypeUpdate::update(circt::firrtl::CircuitOp)::'lambda'(mlir::Operation*)>(long, mlir::Operation*) /home/drom/work/github/llvm/circt/build/../llvm/llvm/include/llvm/ADT/STLExtras.h:179:3
#12 0x00000000006ea438 mlir::detail::walk(mlir::Operation*, llvm::function_ref<mlir::WalkResult (mlir::Operation*)>, mlir::WalkOrder) (.localalias) (../../llvm/circt/build/bin/firtool+0x6ea438)
#13 0x00000000006ea4c9 mlir::detail::walk(mlir::Operation*, llvm::function_ref<mlir::WalkResult (mlir::Operation*)>, mlir::WalkOrder) (.localalias) (../../llvm/circt/build/bin/firtool+0x6ea4c9)
#14 0x00000000006ea4c9 mlir::detail::walk(mlir::Operation*, llvm::function_ref<mlir::WalkResult (mlir::Operation*)>, mlir::WalkOrder) (.localalias) (../../llvm/circt/build/bin/firtool+0x6ea4c9)
#15 0x00000000005c3c3e _ZN4mlir6detail4walkILNS_9WalkOrderE0EZN12_GLOBAL__N_119InferenceTypeUpdate6updateEN5circt6firrtl9CircuitOpEEUlPNS_9OperationEE_S9_NS_10WalkResultEEENSt9enable_ifIXsrN4llvm11disjunctionIJSt7is_sameIT1_S9_ESF_ISG_PNS_6RegionEESF_ISG_PNS_5BlockEEEEE5valueET2_E4typeES9_OT0_ /home/drom/work/github/llvm/circt/build/../llvm/mlir/include/mlir/IR/Visitors.h:135:1
#16 0x00000000005c06b8 mlir::WalkResult mlir::Operation::walk<(mlir::WalkOrder)0, (anonymous namespace)::InferenceTypeUpdate::update(circt::firrtl::CircuitOp)::'lambda'(mlir::Operation*), mlir::WalkResult>((anonymous namespace)::InferenceTypeUpdate::update(circt::firrtl::CircuitOp)::'lambda'(mlir::Operation*)&&) /home/drom/work/github/llvm/circt/build/../llvm/mlir/include/mlir/IR/Operation.h:523:3
#17 0x00000000005bcd5d mlir::WalkResult mlir::OpState::walk<(mlir::WalkOrder)0, (anonymous namespace)::InferenceTypeUpdate::update(circt::firrtl::CircuitOp)::'lambda'(mlir::Operation*), mlir::WalkResult>((anonymous namespace)::InferenceTypeUpdate::update(circt::firrtl::CircuitOp)::'lambda'(mlir::Operation*)&&) /home/drom/work/github/llvm/circt/build/../llvm/mlir/include/mlir/IR/OpDefinition.h:162:3
#18 0x00000000005b7f2c (anonymous namespace)::InferenceTypeUpdate::update(circt::firrtl::CircuitOp) /home/drom/work/github/llvm/circt/build/../lib/Dialect/FIRRTL/Transforms/InferWidths.cpp:1160:18
#19 0x00000000005b9203 (anonymous namespace)::InferWidthsPass::runOnOperation() /home/drom/work/github/llvm/circt/build/../lib/Dialect/FIRRTL/Transforms/InferWidths.cpp:1377:13
#20 0x000000000099daf6 mlir::detail::OpToOpPassAdaptor::run(mlir::Pass*, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int) (../../llvm/circt/build/bin/firtool+0x99daf6)
#21 0x000000000099dc10 mlir::detail::OpToOpPassAdaptor::runPipeline(llvm::iterator_range<llvm::pointee_iterator<std::unique_ptr<mlir::Pass, std::default_delete<mlir::Pass> >*, mlir::Pass> >, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int, mlir::PassInstrumentor*, mlir::PassInstrumentation::PipelineParentInfo const*) (../../llvm/circt/build/bin/firtool+0x99dc10)
#22 0x000000000099d0f4 mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool) (../../llvm/circt/build/bin/firtool+0x99d0f4)
#23 0x000000000099d8da mlir::detail::OpToOpPassAdaptor::run(mlir::Pass*, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int) (../../llvm/circt/build/bin/firtool+0x99d8da)
#24 0x000000000099dc10 mlir::detail::OpToOpPassAdaptor::runPipeline(llvm::iterator_range<llvm::pointee_iterator<std::unique_ptr<mlir::Pass, std::default_delete<mlir::Pass> >*, mlir::Pass> >, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int, mlir::PassInstrumentor*, mlir::PassInstrumentation::PipelineParentInfo const*) (../../llvm/circt/build/bin/firtool+0x99dc10)
#25 0x000000000099e602 mlir::PassManager::runPasses(mlir::Operation*, mlir::AnalysisManager) (../../llvm/circt/build/bin/firtool+0x99e602)
#26 0x000000000099ee3d mlir::PassManager::run(mlir::Operation*) (../../llvm/circt/build/bin/firtool+0x99ee3d)
#27 0x000000000040ccce processBuffer(std::unique_ptr<llvm::MemoryBuffer, std::default_delete<llvm::MemoryBuffer> >, llvm::StringRef, mlir::TimingScope&, mlir::MLIRContext&, std::function<mlir::LogicalResult (mlir::ModuleOp)>) /home/drom/work/github/llvm/circt/build/../tools/firtool/firtool.cpp:277:13
#28 0x000000000040d3f2 executeFirtool(mlir::MLIRContext&) /home/drom/work/github/llvm/circt/build/../tools/firtool/firtool.cpp:357:30
#29 0x000000000040d5cf main /home/drom/work/github/llvm/circt/build/../tools/firtool/firtool.cpp:393:31
#30 0x0000000004c97b35 __libc_start_main (/lib64/libc.so.6+0x27b35)
#31 0x000000000040c0de _start /home/abuild/rpmbuild/BUILD/glibc-2.33/csu/../sysdeps/x86_64/start.S:122:0
./process1.sh: line 16:  1430 Aborted                 (core dumped) valgrind -q ../../llvm/circt/build/bin/firtool a_top_mod.fir --lower-to-hw --imconstprop --infer-widths --lowering-options=noAlwaysFF --mlir-timing --verilog -o=$VFILE2
```

Compiled with `firrtl-1.5-SNAPSHOT` produces this Error:

```
Exception in thread "main" firrtl.passes.CheckWidths$TailWidthException: : [target ~top_mod|top_mod] Parameter 15 in tail operator is larger than input width 14.
```

fabianschuiki, 2021-06-12:
https://github.com/llvm/circt/blob/1e99c0ac2dfc6e77a87fc102051fc38941969568/lib/Dialect/FIRRTL/Transforms/InferWidths.cpp#L1289 🙈

All hail @drom's fuzzer!

seldridge, 2021-06-12:
[image]

fabianschuiki, 2021-06-13:
`firtool` now produces the following:
```plain
/tmp/foo.fir:4:14: error: amount must be less than or equal operand width
    tmp74 <= tail(UInt("b11101001111100"), 15)
```

drom, 2021-06-24:
Strange, I am still getting errors like this.

```
circuit top_mod :
  module top_mod :
    wire tmp74: UInt
    tmp74 <= tail(UInt("b11101001111100"), 15)
```

fabianschuiki, 2021-06-24:
Can confirm that we have regressed on this fron. Looks like a slightly new case, as none of the existing regression tests fail. I'll look into it.

commit f772527f (2021-06-13): [FIRRTL][InferWidths] Infer variables to be 0 or greater

* Fix an issue in width inference where the constraint solver would
  happily assign the solution of a variable's constraint to the variable
  itself. This could be problematic, since the right hand side in a
  constraint `x >= RHS` could be negative in some cases. That would trip
  up the type update. To avoid this, ensure that free variables (which
  correspond to inferrable bit widths in types) are lower-bounded to 0.

* Fixes #1255.
commit 5f140dc1 (2021-06-24): [FIRRTL][InferWidths] Fix vars inferring widths less than 0

- Fix an issue where widths could be inferred as less than zero, which
  would trip up other parts of the system. There was an earlier fix that
  prevented this, but that got accidentally removed in the rewrite of
  the solver/cycle breaker.

- Fixes #1255 (again).
closed 2021-06-24 by fabianschuiki