llvm/circt #1263: InferWidths takes a lot of time even on low FIRRTL inputs

lattner, 2021-06-14:
InferWidths is taking about 15% of runtime even on a low-firrtl input.  It shouldn't have to build a constraint graph or do other fancy things in the obvious cases.  On a large testcase on my laptop, I'm seeing:

```
  Total Execution Time: 29.7302 seconds

  ----User Time----  ----Wall Time----  ----Name----
   14.1005 ( 27.2%)   14.1005 ( 47.4%)  FIR Parser
   28.9724 ( 55.8%)   11.5828 ( 39.0%)  'firrtl.circuit' Pipeline
    4.5414 (  8.7%)    4.5414 ( 15.3%)    InferWidths
    1.2270 (  2.4%)    1.2270 (  4.1%)    LowerFIRRTLTypes
   18.1559 ( 35.0%)    2.7916 (  9.4%)    'firrtl.module' Pipeline
    5.2287 ( 10.1%)    1.2955 (  4.4%)      ExpandWhens
    6.1360 ( 11.8%)    1.1926 (  4.0%)      CSE
...
```

fabianschuiki, 2021-06-14:
Will take a look. We can likely just skip entire modules if they contain no uninferred widths.

fabianschuiki, 2021-06-14:
Improved in c1745d9d. Adds fast paths to skip modules and the entire type update part of `InferWidths` if no uninferred widths are present. Lowers pass time on my machine on fully inferred inputs by 4x to 8x (down to 1-2% overall). `--verify-each=false` is important; otherwise the actual InferWidths pass time gets overshadowed by the verification afterwards.

closed 2021-06-14 by fabianschuiki

lattner, 2021-06-14:
Agreed, the verifier is about to be fixed, xref: https://reviews.llvm.org/D104207

Thank you!