llvm/circt #1270: [FIRRTL] Check Widths False Positive

seldridge, 2021-06-14:
The following circuit (extracted from a real design) is tripping a CheckWidths error:

```scala
circuit Foo :
  module Foo :
    input source: UInt<1>
    input cond: UInt<1>

    wire a_set : UInt<1>
    a_set <= UInt<1>("h0")
    when cond :
      node _a_set_T = dshl(UInt<1>("h1"), source)
      a_set <= _a_set_T
```

The MFC's InferWidths pass is currently erroring out with:
```
fire-drill/TileLinkMonitor.hi.fir:10:13: error: destination width 1 is not greater than or equal to source width 2
      a_set <= _a_set_T
            ^
fire-drill/TileLinkMonitor.hi.fir:10:13: note: see current operation: "firrtl.connect"(%a_set, %1) : (!firrtl.uint<1>, !firrtl.uint<2>) -> ()
```

However, the SFC allows this through. The SFC doesn't align with the spec and does auto-truncation on connects. 🤷‍♀️ 

The eventual SFC output Verilog is:
```
module Foo(
  input   source,
  input   cond
);
  wire [1:0] _a_set_T = 2'h1 << source;
  wire [1:0] _GEN_0 = cond ? _a_set_T : 2'h0; // @[]
  wire  a_set = _GEN_0[0];
endmodule
```

fabianschuiki, 2021-06-15:
This is pretty nasty. I'm a bit nervous about letting connects implicitly truncate bits -- seems at odds with a lot of other FIRRTL design choices. But on the other hand, we also want to be able to process what comes out of Chisel, so that's a bit of a dilemma :see_no_evil:.

Would it make sense to try and fix this on the Chisel side in the long term? Looks like emitting a `dhslw` or a partial connect `<-` could do the trick :thinking:.

For the immediate term, maybe we could add a `--implicit-connect-trunc` option or similar which runs a fixup pass over the IR, inserting a `trunc` op here? That would keep us from actually implementing truncating connects throughout the code base, which we might not want to have in the long run.

seldridge, 2021-06-16:
Something fixed this. The fix seems to be before the last LLVM bump, so I'm not going to bother bisecting it. Closing it for now.

closed 2021-06-16 by seldridge