llvm/circt #1271: [FIRRTL] Width Inference Problem with Cycle

seldridge, 2021-06-14:
The following circuit causes an error during width inference of register `a`. However, the SFC infers a width of two.

```scala
circuit Bar :
  module Bar :
    input clock : Clock
    input cond : UInt<1>

    reg a : UInt, clock

    node _false = add(a, UInt<1>(0))
    node false = tail(_false, 1)
    a <= mux(cond, UInt<2>(0), false)
```

With the MFC, I get:
```
Bar.fir:6:5: error: 'firrtl.reg' op is constrained to be wider than itself
    reg a : UInt, clock
    ^
Bar.fir:6:5: note: see current operation: %a = "firrtl.reg"(%clock) {annotations = [], name = "a"} : (!firrtl.clock) -> !firrtl.uint
Bar.fir:8:19: note: constrained width W >= W+2 here:
    node _false = add(a, UInt<1>(0))
                  ^
Bar.fir:9:18: note: constrained width W >= W+1 here:
    node false = tail(_false, 1)
                 ^
Bar.fir:10:10: note: constrained width W >= W+1 here:
    a <= mux(cond, UInt<2>(0), false)
         ^
```

With the SFC, I get:
```scala
circuit Bar :
  module Bar :
    input clock : Clock
    input cond : UInt<1>

    reg a : UInt<2>, clock with :
      reset => (UInt<1>("h0"), a)
    node _false = add(a, UInt<1>("h0"))
    node false = tail(_false, 1)
    a <= mux(cond, UInt<2>("h0"), false)
```

fabianschuiki, 2021-06-15:
It's likely this will get fixed by tightening the lower bounds on expressions in the cycle breaker (#1269). I'll take a look!

fabianschuiki, 2021-06-15:
Confirmed fixed by #1269. Will add this as a regression test case to that PR.

closed 2021-06-15 by fabianschuiki