llvm/circt #1279: Make width inference lazy

fabianschuiki, 2021-06-15:
The `InferWidths` pass on the FIRRTL dialect greedily constructs a constraint expression tree as it traverses the IR. There already is an optimization which at least lets it skip a module if it contains no uninferred widths at all. However in many cases there are just a handful of widths to be inferred, such that only a small portion of the generated constraint expressions end up being used.

Rather than having such a binary all-or-nothing approach, the `InferWidths` pass could operate lazily by walking a module's IR, and upon encountering an operation that imposes a constraint (`ConnectOp`, `PartialConnectOp`, `Attach`), recursively generate the constraint expression for its operands and memoize the result. This would do expression construction and detection of skippable modules in one go, and would make the pre-pass that ensures modules have constraint variables declared for their ports obsolete.

lattner, 2021-09-03:
Is this still interesting?

fabianschuiki, 2021-09-03:
`InferWidths` has turned out to be pretty quick on most practical examples we've run it on. That doesn't mean we won't see cases where this lazy evaluation will be valuable and needed in the future. But at the moment it hasn't bubbled up the priority stack.

lattner, 2021-09-03:
k, I'd recommend closing this issue.  I don't this it is serving much purpose

fabianschuiki, 2021-09-03:
Done

closed 2021-09-03 by fabianschuiki