llvm/circt #1281: Support width annotations

fabianschuiki, 2021-06-15:
Add support for bit width annotations in the `InferWidths` pass of the FIRRTL dialect.

lattner, 2021-09-03:
Is this important?

fabianschuiki, 2021-09-03:
I haven't seen these annotations in any of our internal code yet, so it doesn't seem very urgent. But there are tests in the Scala code for this and I could totally see someone debugging a Chisel design and throwing in some of these annotations to test if something works.

lattner, 2021-09-03:
How about detecting these and emitting an error if they appear, we can then close out this bug.  If someone needs it in the future we can tackle it then.

fabianschuiki, 2021-09-03:
If you run firtool with `--warn-on-unprocessed-annotations`, you'll get a warning/error for these. But that's opt-in. Are you proposing to make this a hard error as part of `InferWidths`? That would totally make sense as well.

lattner, 2021-09-03:
I don't know how unimportant this is.  If it is something we know about that could plausibly come up, then catching it proactively seems like a good thing.

I mostly just want to clear out this bug if we don't plan to do anything about it.

fabianschuiki, 2021-09-03:
I'll check SiFive's internal usage of this annotation. If it isn't really used, I suggest to do what you say and just error out when we see these annos. If we do make use of them, I'll push this forward.