llvm/circt #1282: [comb] Investigate porting XLS inspired optimizations

lattner, 2021-06-15:
The XLS project has a [number of nice optimizations](https://google.github.io/xls/optimizations/), we should look to see if any make sense to implement at the comb level.

youngar, 2021-08-21:
Since we have some KnownBits analysis coming in to the canonicalizers, I wanted to ask about the first section of the XLS optimizations, "Narrowing Operations".  They advocate for moving known bits out of computations.

This example removes the known bits of the operands to the comparison:
```mlir
// `x` and `y` both have known bits:
// zeros: 0000
// ones:  0001

%0 = eq %x, %y

->

%0 = extract(%x, 1)
%1 = extract(%y, 1)
%2 = eq %0, %1
```
I see how this is helpful in the HW cost model, but I wonder if these optimizations (if aggressively added) would make the resulting verilog harder to read.  Should we add these optimizations (or have we already started)?  Can the synthesis tools reliably handle this kind of thing for us?

lattner, 2021-08-24:
I agree that there is a balance to strike and we need to watch out for this in the future.  I'll also be revising this, because I think that "using known bits in canonicalize" is actually an antipattern in LLVM that burns compile time, and I have a better idea for how to handle it.

That said, the case you point out seems like a clear with w.r.t. readability to me, because you're changing `(thing1 & big_constant) == (thing2 & otherbig_constant)` to something like `thing1[3:1] != thing2[3:1]`.  The place where it isn't great for readability is if one equality gets turned into a pile of equality comparisons.  I have a plan for how to handle this, I'll play with it when I revise the `MaskedBits` stuff into its own pass.