llvm/circt #1283: [FIRRTL] Apparently huge memory regression in Chipyard test

lattner, 2021-06-15:
The chart at the [bottom of this page](https://circt.org/perf/) shows a 70M memory regression, we should characterize this.

prithayan, 2021-06-16:
The memory regression seems to be due to the Disabling of multi threaded parser, https://github.com/llvm/circt/commit/b6737713adb1fbd82e2afc6a7da1bce49f322e95

lattner, 2021-06-16:
Whoa, really?  That's surprising to me.

lattner, 2021-06-16:
How is memory measured in this case?  One change the patch introduces is that it would cause multiple instances of the verifier to run in parallel right after parsing the .fir file.  The verifier doesn't use a ton of memory though, just has a transient DominanceSet (which we probably don't even populate)

prithayan, 2021-06-18:
I tried once again, https://github.com/llvm/circt/commit/ca1831b38d8f56cb792b0f4618812400c649f12a vs https://github.com/llvm/circt/commit/b6737713adb1fbd82e2afc6a7da1bce49f322e95 using `/usr/bin/time -v` tool to measure the max heap usage. 
which is `Maximum resident set size (kbytes): 802396`  vs `Maximum resident set size (kbytes): 881396` .
I used `valgrind --tool=massif` and it seems the peak usage is in the parser, here's a sample from the output showing the heap allocation and the call stack. The number in brackets is the bytes allocated. I'm going to dig deeper to understand it better. 
After enabling multithreading:: 
  ```95.94% (71,471,205B) (heap allocation functions) malloc/new/new[], --alloc-fns, etc.
->41.55% (30,951,864B) 0x770AFD: mlir::Operation::create(mlir::Location, mlir::OperationName, mlir::TypeRange, mlir::ValueRange, mlir::DictionaryAttr, mlir::BlockRange, unsigned int) (mlir/lib/IR/Operation.cpp:136)
| ->41.55% (30,951,864B) 0x771045: mlir::Operation::create(mlir::Location, mlir::OperationName, mlir::TypeRange, mlir::ValueRange, mlir::DictionaryAttr, mlir::BlockRange, mlir::RegionRange) (mlir/lib/IR/Operation.cpp:94)
|   ->41.55% (30,951,864B) 0x770F52: mlir::Operation::create(mlir::OperationState const&) (mlir/lib/IR/Operation.cpp:83)
|     ->41.55% (30,951,864B) 0x6BA092: mlir::OpBuilder::createOperation(mlir::OperationState const&) (mlir/lib/IR/Builders.cpp:400)
|       ->14.05% (10,470,144B) 0x55A4BD: circt::firrtl::SkipOp mlir::OpBuilder::create<circt::firrtl::SkipOp>(mlir::Location) (in /home/prithayanb/release/circt/build/bin/firtool)
|       | ->14.05% (10,470,144B) 0x547911: (anonymous namespace)::FIRStmtParser::parseSkip() (in /home/prithayanb/release/circt/build/bin/firtool)
|       |   ->14.05% (10,470,144B) 0x5416AC: (anonymous namespace)::FIRStmtParser::parseSimpleStmt(unsigned int) (in /home/prithayanb/release/circt/build/bin/firtool)
|       |     ->14.05% (10,470,144B) 0x53EE53: (anonymous namespace)::FIRStmtParser::parseSimpleStmtBlock(unsigned int) (in /home/prithayanb/release/circt/build/bin/firtool)
|       |       ->14.05% (10,470,144B) 0x53BF2D: circt::firrtl::importFIRRTL(llvm::SourceMgr&, mlir::MLIRContext*, circt::firrtl::FIRParserOptions) (in /home/prithayanb/release/circt/build/bin/firtool)
|       |         ->14.05% (10,470,144B) 0x414AD7: executeFirtool(mlir::MLIRContext&) (in /home/prithayanb/release/circt/build/bin/firtool)
|       |           ->14.05% (10,470,144B) 0x41433D: main (in /home/prithayanb/release/circt/build/bin/firtool)
|       |

```
Before enabling Multithreading::
```40.71% (21,312,312B) 0x770A9D: mlir::Operation::create(mlir::Location, mlir::OperationName, mlir::TypeRange, mlir::ValueRange, mlir::DictionaryAttr, mlir::BlockRange, unsigned int) (mlir/lib/IR/Operation.cpp:136)
| ->40.71% (21,312,312B) 0x770FE5: mlir::Operation::create(mlir::Location, mlir::OperationName, mlir::TypeRange, mlir::ValueRange, mlir::DictionaryAttr, mlir::BlockRange, mlir::RegionRange) (mlir/lib/IR/Operation.cpp:94)
|   ->40.71% (21,312,312B) 0x770EF2: mlir::Operation::create(mlir::OperationState const&) (mlir/lib/IR/Operation.cpp:83)
|     ->40.71% (21,312,312B) 0x6BA032: mlir::OpBuilder::createOperation(mlir::OperationState const&) (mlir/lib/IR/Builders.cpp:400)
|       ->14.68% (7,686,976B) 0x55A45D: circt::firrtl::SkipOp mlir::OpBuilder::create<circt::firrtl::SkipOp>(mlir::Location) (in /home/prithayanb/release/circt/build/bin/firtool)
|       | ->14.68% (7,686,976B) 0x5478B1: (anonymous namespace)::FIRStmtParser::parseSkip() (in /home/prithayanb/release/circt/build/bin/firtool)
|       |   ->14.68% (7,686,976B) 0x54164C: (anonymous namespace)::FIRStmtParser::parseSimpleStmt(unsigned int) (in /home/prithayanb/release/circt/build/bin/firtool)
|       |     ->14.68% (7,686,976B) 0x53EDF3: (anonymous namespace)::FIRStmtParser::parseSimpleStmtBlock(unsigned int) (in /home/prithayanb/release/circt/build/bin/firtool)
|       |       ->14.68% (7,686,976B) 0x53BECD: circt::firrtl::importFIRRTL(llvm::SourceMgr&, mlir::MLIRContext*, circt::firrtl::FIRParserOptions) (in /home/prithayanb/release/circt/build/bin/firtool)
|       |         ->14.68% (7,686,976B) 0x414AFA: executeFirtool(mlir::MLIRContext&) (in /home/prithayanb/release/circt/build/bin/firtool)
|       |           ->14.68% (7,686,976B) 0x41433D: main (in /home/prithayanb/release/circt/build/bin/firtool)
```

lattner, 2021-06-19:
Super weird.  Enabling/disabling multithreading doesn't change the IR that we create (because the parser isn't actually threaded).

The only thing I see that could be weird in MLIRContext is in `Identifier::get`.  It seems to have a "localIdentifierCache" that is only used in threaded mode.  I wonder if that is part of the problem, and I wonder how it reacts to changes in the multithreading on/off change.

lattner, 2021-06-22:
Does fedf6728 fix this?  

The per-thread identifier cache seems poorly considered to me, but that patch generally reduces load on the identifier table.

prithayan, 2021-06-22:
Nice ! It does fix the memory regression, 
Max resident set size reduced ~70M,  from 880688 KB to 829584 KB after the checkin.

closed 2021-06-22 by lattner

lattner, 2021-06-22:
I filed https://bugs.llvm.org/show_bug.cgi?id=50802 to track whether ThreadLocalCache in MLIRContext is a good idea in the first place.