llvm/circt #1288: [FIRRTL] (dshl, head) mismatch

drom, 2021-06-16:
The following FIRRTL program

```python
; seed: 759
circuit top_mod :
  module top_mod :
    input inp_cb: UInt<60>
    output _tmp66: UInt<91>
    wire tmp12: SInt
    wire _tmp13: UInt
    wire tmp58: UInt
    tmp12 <= shl(SInt("b-100101100101"), 1)
    _tmp13 <= dshl(inp_cb, tail(asUInt(tmp12), 11))
    tmp58 <= pad(inp_cb, 41)
    _tmp66 <= dshl(_tmp13, head(tmp58, 0))
```

Compiled with `firtool` produces this Verilog:

```verilog
module top_mod(
  input  [59:0] inp_cb,
  output [90:0] _tmp66);

  assign _tmp66 = {24'h0, {1'h0, inp_cb, 6'h0} << 67'h0};	// a_top_mod.fir:3:3, :10:15, :12:15
endmodule
```

Compiled with `firrtl-1.5-SNAPSHOT` produces this Verilog:

```verilog
module top_mod(
  input  [59:0] inp_cb,
  output [90:0] _tmp66
);
  wire [65:0] _GEN_0 = {inp_cb, 6'h0};
  wire [66:0] _GEN_1 = {{1'd0}, _GEN_0};
  wire [59:0] _GEN_2 = {{1'd0}, _GEN_1[58:0]};
  assign _tmp66 = {{31'd0}, _GEN_2};
endmodule
```

`Yosys 0.9+3911 (git sha1 4b31223e, clang 11.0.1 -fPIC -Os)` reports formal mismatch:

```
ERROR: Found 7 unproven $equiv cells in 'equiv_status -assert'.
```

fabianschuiki, 2021-06-16:
SFC infers the widths as follows:
```firrtl
circuit top_mod:
  module top_mod:
    input inp_cb: UInt<60>
    output _tmp66: UInt<91>
    wire tmp12: SInt<14>
    wire _tmp13: UInt<59>
    wire tmp58: UInt<60>
    tmp12 <= shl(SInt<13>("h-965"), 1)
    _tmp13 <= dshl(inp_cb, tail(asUInt(tmp12), 11))
    tmp58 <= pad(inp_cb, 41)
    _tmp66 <= dshl(_tmp13, head(tmp58, 0))
```
Interesting is `_tmp13`, which is inferred to be 59 bits wide. The `dshl` is passed `inp_cb`, which is 60, and `tail(asUInt(tmp12), 11)`, which technically should be 3 bits wide (`tmp12` is 14, which leaves a tail of 3 starting at 11 bits). MFC infers `_tmp13` as 67, which seems correct to me (`60 + 2^3 - 1`). The `dshl` has a `IsPow(RHS)` in its `propagateType` implementation, which here seems to evaluate to 0.

Related to #1289? @azidar?

drom, 2021-07-12:
This is FIRTOOL output with don't touch:

```
module top_mod(
  input  [59:0] inp_cb,
  output [90:0] _tmp66);

  wire [13:0] tmp12;	// a_top_mod.fir:6:5
  wire [66:0] _tmp13;	// a_top_mod.fir:7:5
  wire [59:0] tmp58;	// a_top_mod.fir:8:5

  assign tmp12 = 14'h2D36;	// a_top_mod.fir:9:{11,14}
  wire [13:0] _T = tmp12;	// a_top_mod.fir:10:28
  assign _tmp13 = {7'h0, inp_cb} << {64'h0, _T[2:0]};	// a_top_mod.fir:10:{12,15,28}
  assign tmp58 = inp_cb;	// a_top_mod.fir:11:11
  assign _tmp66 = {24'h0, _tmp13 << 67'h0};	// a_top_mod.fir:3:10, :12:15
endmodule
```

lattner, 2021-07-13:
It's be great to analyze this if you haven't already Fabian, this looks like a possible subtle miscompilation.

fabianschuiki, 2021-07-14:
This looks correct to me -- I think MFC is doing the right thing in both cases. Both MFC outputs are equivalent according to yosys:
```
Found 91 $equiv cells in equiv:
  Of those cells 91 are proven and 0 are unproven.
  Equivalence successfully proven!
```
Pretty sure this is a SFC bug. Quick rundown of the rationale:

```
wire tmp12: SInt
tmp12 <= shl(SInt("b-100101100101"), 1)
```
The literal is `-0x965`, which in two's complement is `0x169b`, requiring 13 bits to be represented. Shifted by 1 to the left this is `0x2d36`, requiring 14 bits to be represented. The Verilog reflects this:
```verilog
wire [13:0] tmp12;
assign tmp12 = 14'h2D36;
```
---
```
wire _tmp13: UInt
_tmp13 <= dshl(inp_cb, tail(asUInt(tmp12), 11))
```
There's a lot going on. `asUInt(tmp12)` is the `0x2d36` at 14 bits from above. `tail(..., 11)` cuts off the front 11 bits, leaving the last 3 bits of `tmp12`. This is reflected in the Verilog:
```verilog
wire [13:0] _T = tmp12;
... << {..., _T[2:0]};
```
The `dshl(inp_cb, ...)` extends the width of `inp_cb` by the maximum value that the RHS can hold. The RHS is the `tail(...)` above with 3 bits, which adds 7 (2^3-1) to the width of `inp_cb`, resulting in 67 bits. This is reflected in the Verilog, where both LHS and RHS of the shift are extended to 67 bits, and the `_tmp13` wire gets the right size:
```verilog
wire [66:0] _tmp13;
assign _tmp13 = {7'h0, inp_cb} << {64'h0, _T[2:0]};
```
---
```
wire tmp58: UInt
tmp58 <= pad(inp_cb, 41)
```
The padding here is a no-op, because 41 is smaller than `inp_cb`'s width of 60 bits, so the resulting `tmp58` is 60 bits. The Verilog just does an assign here:
```verilog
wire [59:0] tmp58;
assign tmp58 = inp_cb;
```
---
```
_tmp66 <= dshl(_tmp13, head(tmp58, 0))
```
This one is subtle: the `head(tmp58, 0)` extracts the `0` most significant bits of `tmp58`, which is a value of 0 bits. During lowering to the HW dialect, `head(..., 0)` is specifically mapped to a constant zero. The Verilog reflects this:
```verilog
... = {... << 67'h0};
```
Since the `head(...)` is 0 bits wide, the `dshl` has the same 67 bits as its LHS `_tmp13`. However, since we're assigning this 67 bit value to `_tmp66` which is 91 bits wide, the result of the shift is zero padded by 24 bits (91-67). The Verilog reflects this:
```verilog
assign _tmp66 = {24'h0, _tmp13 << 67'h0};
```

lattner, 2021-07-14:
Great, should this be closed then and moved to SFC?

fabianschuiki, 2021-07-16:
Totally. Opened https://github.com/chipsalliance/firrtl/issues/2296 to track this further.

closed 2021-07-16 by fabianschuiki