llvm/circt #1289: [FIRRTL] (dshl, tail, mul) width inference mismatch

drom, 2021-06-16:
The following FIRRTL program

Produces different width for the output port `_tmp40`

```python
; seed: 1144
circuit top_mod :
  module top_mod :
    output _tmp40: UInt
    wire tmp19: UInt
    tmp19 <= mul(UInt("o20575"), UInt<1>(0))
    _tmp40 <= dshl(UInt("b10011010000001011101"), tail(tmp19, 15))
```

Compiled with `firtool` produces this Verilog:

```verilog
module top_mod(
  output [19:0] _tmp40);

  assign _tmp40 = 20'h9A05D;	// a_top_mod.fir:3:3, :7:20
endmodule
```

Compiled with `firrtl-1.5-SNAPSHOT` produces this Verilog:

```verilog
module top_mod(
  output [18:0] _tmp40
);
  assign _tmp40 = 19'h1a05d;
endmodule
```

`Yosys 0.9+3911 (git sha1 4b31223e, clang 11.0.1 -fPIC -Os)` reports formal mismatch:

```
ERROR: Can't match gold port `_tmp40_gold' to a gate port.
```

fabianschuiki, 2021-06-16:
This is very weird. SFC seems to truncate the uppermost bit of the constant `UInt("b10011010000001011101")`.

MFC infers the widths as:
```firrtl
circuit top_mod :
  module top_mod :
    output _tmp40 : UInt<20>  ; <--- difference
    wire tmp19 : UInt<15>
    tmp19 <= mul(UInt<14>("h217d"), UInt<1>("h0"))
    _tmp40 <= dshl(UInt<20>("h9a05d"), tail(tmp19, 15))
```

SFC infers the widths as:
```firrtl
circuit top_mod :
  module top_mod :
    output _tmp40 : UInt<19>  ; <--- difference
    wire tmp19 : UInt<15>
    tmp19 <= mul(UInt<14>("h217d"), UInt<1>("h0"))
    _tmp40 <= dshl(UInt<20>("h9a05d"), tail(tmp19, 15))
```

The only difference is the width of the output. Seems like `dshl` is inferred to have width 19 with the given operands in SFC. That might happen because of the LHS being `UInt<20>`, and the RHS is `UInt<0>`. In SFC, `propagateType` of `Dshl` says:
```scala
  case object Dshl extends PrimOp {
    override def propagateType(e: DoPrim): Type = t1(e) match {
      case _: UIntType  => UIntType(IsAdd(w1(e), IsAdd(IsPow(w2(e)), Closed(-1))))
      // [...]
    }
    // [...]
  }
```
That would explain the result: `IsAdd(w1(e), IsAdd(IsPow(w2(e)), Closed(-1)))` translates to `IsAdd(20, IsAdd(IsPow(?), -1))`. That would result in 19 if the `IsPow(0)` would be treated as `0` somehow; or if the width of `tail(tmp19, 15)` some how turned out to be `-1` (because the `mul` might fold to just a zero constant of the LHS, which is one bit less than what the `mul` would yield). Maybe @azidar knows what's up here.

drom, 2021-06-16:
Similar issue:

The following FIRRTL program

Produces different width for the output port `_tmp40`

```python
; seed: 1337
circuit top_mod :
  module top_mod :
    input inp_e: SInt<3>
    output tmp37: SInt
    output tmp23: UInt
    tmp23 <= UInt<15>("b100101001010011")
    tmp37 <= dshl(inp_e, tail(tmp23, 7))
```

Compiled with `firtool` produces this Verilog:

```verilog
module top_mod(
  input  [2:0]   inp_e,
  output [257:0] tmp37,
  output [14:0]  tmp23);

  assign tmp37 = {{255{inp_e[2]}}, inp_e} << 258'h53;	// a_top_mod.fir:3:3, :8:14
  assign tmp23 = 15'h4A53;	// a_top_mod.fir:3:3, :7:14
endmodule
```

Compiled with `firrtl-1.5-SNAPSHOT` produces this Verilog:

```verilog
module top_mod(
  input  [2:0]  inp_e,
  output [1:0]  tmp37,
  output [14:0] tmp23
);
  wire [257:0] _GEN_0 = {{255{inp_e[2]}},inp_e};
  wire [257:0] _GEN_2 = $signed(_GEN_0) << tmp23[7:0];
  assign tmp37 = _GEN_2[1:0];
  assign tmp23 = 15'h4a53;
endmodule
```

`Yosys 0.9+3911 (git sha1 4b31223e, clang 11.0.1 -fPIC -Os)` reports formal mismatch:

```
ERROR: Can't match gold port `_tmp40_gold' to a gate port.
```