llvm/circt #1297: [FIRRTL] Merge CheckWidths into InferWidths

fabianschuiki, 2021-06-17:
Merge the `CheckWidths` pass into the `InferWidths` pass. This should be fairly easy to do, since during the *type update* phase of width inference we're already visiting all the parts of the IR that are interesting. We can just run the checks that `CheckWidths` is doing there as well (or actually fold them into the type updating procedure), and emit errors appropriately.

lattner, 2021-06-17:
Isn't the idea of CheckWidths to be an independent thing that is run later?

fabianschuiki, 2021-06-17:
I've only ever seen `CheckWidths` run immediately after `InferWidths` in the Scala implementation. More specifically, on the Scala side the inference code does not generate any errors at all, and expects these to be reported later in the check pass. This seems like a needless re-iteration over the IR, especially since we already report inference issues in the `InferWidths` pass where we have all the context info *why* inference failed. Seems weird to have a pass say "cannot infer this because X is Y and Z is bad", and then another pass say "uninferred here".

lattner, 2021-06-17:
Ok, then yes, I agree this shouldn't be a pass.  It's effectively an assertion in CheckWidths.

PR #1422 (2021-07-16): [FIRRTL] Merge CheckWidths into InferWidths
Merge the `CheckWidhts` pass into the `InferWidths` pass, which already has all the necessary information to complain to the user about uninferred widths. This commit also improves error reporting a bit by tracking additional location information on the constraint expression such that we can point the user at the sites where a connection may cause problems.

Fixes #1297.
commit b0a6986b (2021-07-16): [FIRRTL] Merge CheckWidths into InferWidths

Merge the `CheckWidhts` pass into the `InferWidths` pass, which already
has all the necessary information to complain to the user about
uninferred widths. This commit also improves error reporting a bit by
tracking additional location information on the constraint expression
such that we can point the user at the sites where a connection may
cause problems.

Fixes #1297.
closed 2021-07-16 by fabianschuiki