llvm/circt #1299: [PyCDE] Graphing API failing

mikeurbach, 2021-06-17:
The test that exercises this is failing for me: https://github.com/llvm/circt/blob/1ac87078f7ff95827af1ad3777e0bc3d4d7736e1/frontends/PyCDE/src/pycde/system.py#L46-L50

Specifically, I get this error: 

```
MLIR Textual PassPipeline Parser:1:1: error: 'view-op-graph' does not refer to a registered pass or pass pipeline
```

I know that pass is part of `mlir-opt`, and I would have expected `import mlir.all_passes_registration` to make it available. Is there some other step I'm missing? Perhaps a CMake dependency that wasn't declared? I've built the following targets: `MLIRBindingsPythonExtension`, `CIRCTBindingsPython`, `PyCDE`, and `check-pycde`.

teqdruid, 2021-06-18:
Maybe LLVM looks for graphviz at config time and didn't find it...?

mikeurbach, 2021-06-18:
Hmm, well I do have the graphviz tools (like `dot`) available. I re-ran CMake just to confirm, and I still have this issue. I'll keep digging into it.

teqdruid, 2021-06-18:
It may build against libgraphviz-dev.

mikeurbach, 2021-07-15:
After some time, and re-configuring CMake, and re-compiling everything, I am no longer hitting this issue.

closed 2021-07-15 by mikeurbach