llvm/circt #1300: FIRRTL dialect annotation support isn't right

lattner, 2021-06-18:
Various FIRRTL dialect ops have an "annotations" attribute which is a required attribute, but they are declared like this in tblgen: `DefaultValuedAttr<AnnotationArrayAttr, "{}">:$annotations);`

This is a problem: it is a required attribute, but defaults to null.  The pretty printers work around this in various ways, but the generic op printer crashes (and other things will probably be unhappy as well):

`circt-translate -import-firrtl -verify-diagnostics test/Dialect/FIRRTL/parse-basic.fir --mlir-print-op-generic`

I think we should move to the design where these are required and empty if not present (we should just change all the build methods to create an empty array with `$ctx`.

One other oddity is that the Module ops don't elide printing the annotations array when empty.

When this is fixed, some defensive null checking code in this patch can be removed: https://github.com/llvm/circt/commit/e8065e6f3c30dc2ccaceadb2fcc31ed0fde1c485

lattner, 2021-06-18:
A complexity here is that default initializers don't get rewritten and can't have $_ctx or $_builder in them.

lattner, 2021-06-18:
Progress: 6ec64886

lattner, 2021-06-18:
Finished in b2f4db87

closed 2021-06-18 by lattner

seldridge, 2021-06-19:
Thanks for cleaning up my code! This looks _way_ better after your patch series.