llvm/circt #1307: [FIRRTL] "sink not fully initialized" error is too verbose

lattner, 2021-06-22:
This testcase:

```
circuit stmts : 
  module stmts :
    input reset : UInt<1>         ; CHECK: in %reset: !firrtl.uint<1>,

   wire _t : UInt<1>[12] @[Nodes.scala 370:76]
   wire _t_2 : UInt<1>[12]

   when reset :
      node n4 = _t_2
      _t <= n4
   else :
      node n4 = _t_2   ; 'n4' name is in unique scopes.
      _t <- n4
```
produces 12 error messages, it should only produce one or two.

```
$ firtool x.fir
x.fir:6:4: error: sink "_t_2_0" not fully initialized
   wire _t_2 : UInt<1>[12]
   ^
x.fir:6:4: note: see current operation: %_t_2_0 = "firrtl.wire"() {annotations = [], name = "_t_2_0"} : () -> !firrtl.uint<1>
x.fir:6:4: error: sink "_t_2_1" not fully initialized
   wire _t_2 : UInt<1>[12]
   ^
x.fir:6:4: note: see current operation: %_t_2_1 = "firrtl.wire"() {annotations = [], name = "_t_2_1"} : () -> !firrtl.uint<1>
x.fir:6:4: error: sink "_t_2_2" not fully initialized
   wire _t_2 : UInt<1>[12]
   ^
x.fir:6:4: note: see current operation: %_t_2_2 = "firrtl.wire"() {annotations = [], name = "_t_2_2"} : () -> !firrtl.uint<1>
x.fir:6:4: error: sink "_t_2_3" not fully initialized
   wire _t_2 : UInt<1>[12]
   ^
x.fir:6:4: note: see current operation: %_t_2_3 = "firrtl.wire"() {annotations = [], name = "_t_2_3"} : () -> !firrtl.uint<1>
x.fir:6:4: error: sink "_t_2_4" not fully initialized
   wire _t_2 : UInt<1>[12]
   ^
x.fir:6:4: note: see current operation: %_t_2_4 = "firrtl.wire"() {annotations = [], name = "_t_2_4"} : () -> !firrtl.uint<1>
x.fir:6:4: error: sink "_t_2_5" not fully initialized
   wire _t_2 : UInt<1>[12]
   ^
x.fir:6:4: note: see current operation: %_t_2_5 = "firrtl.wire"() {annotations = [], name = "_t_2_5"} : () -> !firrtl.uint<1>
x.fir:6:4: error: sink "_t_2_6" not fully initialized
   wire _t_2 : UInt<1>[12]
   ^
x.fir:6:4: note: see current operation: %_t_2_6 = "firrtl.wire"() {annotations = [], name = "_t_2_6"} : () -> !firrtl.uint<1>
x.fir:6:4: error: sink "_t_2_7" not fully initialized
   wire _t_2 : UInt<1>[12]
   ^
x.fir:6:4: note: see current operation: %_t_2_7 = "firrtl.wire"() {annotations = [], name = "_t_2_7"} : () -> !firrtl.uint<1>
x.fir:6:4: error: sink "_t_2_8" not fully initialized
   wire _t_2 : UInt<1>[12]
   ^
x.fir:6:4: note: see current operation: %_t_2_8 = "firrtl.wire"() {annotations = [], name = "_t_2_8"} : () -> !firrtl.uint<1>
x.fir:6:4: error: sink "_t_2_9" not fully initialized
   wire _t_2 : UInt<1>[12]
   ^
x.fir:6:4: note: see current operation: %_t_2_9 = "firrtl.wire"() {annotations = [], name = "_t_2_9"} : () -> !firrtl.uint<1>
x.fir:6:4: error: sink "_t_2_10" not fully initialized
   wire _t_2 : UInt<1>[12]
   ^
x.fir:6:4: note: see current operation: %_t_2_10 = "firrtl.wire"() {annotations = [], name = "_t_2_10"} : () -> !firrtl.uint<1>
x.fir:6:4: error: sink "_t_2_11" not fully initialized
   wire _t_2 : UInt<1>[12]
   ^
x.fir:6:4: note: see current operation: %_t_2_11 = "firrtl.wire"() {annotations = [], name = "_t_2_11"} : () -> !firrtl.uint<1>

```

youngar, 2021-06-22:
Should this just bail after printing 1 error? Since this is running after lower-types each element looks like a different declaration.

lattner, 2021-06-22:
Yeah, that would make sense

youngar, 2021-06-22:
fixed in https://github.com/llvm/circt/pull/1311

closed 2021-06-22 by youngar