llvm/circt #1312: [FIRRTL] Windows build broken

teqdruid, 2021-06-22:
```
D:\a\circt\circt\lib\Dialect\FIRRTL\Import\FIRTokenKinds.def(132,1): error C2039: 'lp_': is not a member of 'circt::firrtl::FIRToken' (compiling source file D:\a\circt\circt\lib\Dialect\FIRRTL\Import\FIRParser.cpp) [D:\a\circt\circt\build_release\lib\Dialect\FIRRTL\Import\obj.CIRCTImportFIRRTL.vcxproj]
D:\a\circt\circt\lib\Dialect\FIRRTL\Import\FIRLexer.h(29): message : see declaration of 'circt::firrtl::FIRToken' (compiling source file D:\a\circt\circt\lib\Dialect\FIRRTL\Import\FIRParser.cpp) [D:\a\circt\circt\build_release\lib\Dialect\FIRRTL\Import\obj.CIRCTImportFIRRTL.vcxproj]
D:\a\circt\circt\lib\Dialect\FIRRTL\Import\FIRTokenKinds.def(132,1): error C2065: 'lp_': undeclared identifier (compiling source file D:\a\circt\circt\lib\Dialect\FIRRTL\Import\FIRParser.cpp) [D:\a\circt\circt\build_release\lib\Dialect\FIRRTL\Import\obj.CIRCTImportFIRRTL.vcxproj]
D:\a\circt\circt\lib\Dialect\FIRRTL\Import\FIRTokenKinds.def(132,1): error C2059: syntax error: ':' (compiling source file D:\a\circt\circt\lib\Dialect\FIRRTL\Import\FIRParser.cpp) [D:\a\circt\circt\build_release\lib\Dialect\FIRRTL\Import\obj.CIRCTImportFIRRTL.vcxproj]
D:\a\circt\circt\lib\Dialect\FIRRTL\Import\FIRTokenKinds.def(155,1): error C2039: 'lp_': is not a member of 'circt::firrtl::FIRToken' (compiling source file D:\a\circt\circt\lib\Dialect\FIRRTL\Import\FIRParser.cpp) [D:\a\circt\circt\build_release\lib\Dialect\FIRRTL\Import\obj.CIRCTImportFIRRTL.vcxproj]
D:\a\circt\circt\lib\Dialect\FIRRTL\Import\FIRLexer.h(29): message : see declaration of 'circt::firrtl::FIRToken' (compiling source file D:\a\circt\circt\lib\Dialect\FIRRTL\Import\FIRParser.cpp) [D:\a\circt\circt\build_release\lib\Dialect\FIRRTL\Import\obj.CIRCTImportFIRRTL.vcxproj]
D:\a\circt\circt\lib\Dialect\FIRRTL\Import\FIRTokenKinds.def(156,1): error C2039: 'lp_': is not a member of 'circt::firrtl::FIRToken' (compiling source file D:\a\circt\circt\lib\Dialect\FIRRTL\Import\FIRParser.cpp) [D:\a\circt\circt\build_release\lib\Dialect\FIRRTL\Import\obj.CIRCTImportFIRRTL.vcxproj]
D:\a\circt\circt\lib\Dialect\FIRRTL\Import\FIRLexer.h(29): message : see declaration of 'circt::firrtl::FIRToken' (compiling source file D:\a\circt\circt\lib\Dialect\FIRRTL\Import\FIRParser.cpp) [D:\a\circt\circt\build_release\lib\Dialect\FIRRTL\Import\obj.CIRCTImportFIRRTL.vcxproj]
D:\a\circt\circt\lib\Dialect\FIRRTL\Import\FIRTokenKinds.def(164,1): error C2039: 'lp_': is not a member of 'circt::firrtl::FIRToken' (compiling source file D:\a\circt\circt\lib\Dialect\FIRRTL\Import\FIRParser.cpp) [D:\a\circt\circt\build_release\lib\Dialect\FIRRTL\Import\obj.CIRCTImportFIRRTL.vcxproj]
D:\a\circt\circt\lib\Dialect\FIRRTL\Import\FIRLexer.h(29): message : see declaration of 'circt::firrtl::FIRToken' (compiling source file D:\a\circt\circt\lib\Dialect\FIRRTL\Import\FIRParser.cpp) [D:\a\circt\circt\build_release\lib\Dialect\FIRRTL\Import\obj.CIRCTImportFIRRTL.vcxproj]
```

lattner, 2021-06-22:
Super weird, the MSVC preprocessor is failing to token paste?  Did something change in the build settings?

teqdruid, 2021-06-22:
Not so far as I can tell:

[image]
[image]

teqdruid, 2021-06-29:
Any progress on this?

teqdruid, 2021-07-13:
@lattner Is there someone at SiFive I can assign to this? The Windows build has been failing every night for 3 weeks now and it's important to keep things from degrading further.

lattner, 2021-07-13:
hi John, thanks for pinging this.  I don't have access to a windows machine to diagnose this.  What I can tell from the log is that the preprocessor is failing to token paste `lp_ ## and` in FIRParser.cpp:1585:

```
#define TOK_LPKEYWORD(SPELLING) case FIRToken::lp_##SPELLING:
#include "FIRTokenKinds.def"
```

The only thing I can think that would cause a regression here is a change in the cmake flags.  "and" is a keyword in C++ and MSVC is apparently failing to paste this.  Do you know what patch started this failing?

teqdruid, 2021-07-14:
This is the first failing build: https://github.com/llvm/circt/actions/runs/960552543
So it was one of the ones on June 21st: https://github.com/llvm/circt/commits/71e9305ef3f22bc4f333b3c98f525289f9f8ced7

It's a nightly build, so that's as close as I can get without bisection testing.

youngar, 2021-07-14:
The windows build seems just kind of generally borked (old and current):
```
CMake Error at llvm/llvm/cmake/modules/AddLLVM.cmake:1791 (add_dependencies):
  The dependency target "llhd-sim" of target "check-circt" does not exist.
... snip ...
-- Generating done
CMake Generate step failed.  Build files cannot be regenerated correctly.
```
Possibly it doesn't make sense to compile after this error?

youngar, 2021-07-14:
I'll tackle that llhd-sim error, started some work here: https://github.com/llvm/circt/compare/main...youngar:windows?expand=1

teqdruid, 2021-07-14:
Did that error pop up with the first broken build or has it been there on the old passing builds?

The llhd-sim binary doesn't link on Windows, so it is disabled intentionally.

youngar, 2021-07-14:
It has been there this whole time

teqdruid, 2021-07-15:
@lattner FYI: the issue which @youngar noticed was fixed by #1425. This issue persists, as expected.

youngar, 2021-07-15:
Did some more investigation of the problem here: https://github.com/llvm/circt/pull/1430. I have a code change that appears to fix the Windows build, but I don't understand why.

youngar, 2021-07-16:
#1430 is merged and the windows build is green again!  The resolution is a little sad, it would be better to not have a hacky work-around, but I think we can close this.

closed 2021-07-16 by youngar