llvm/circt #1313: [FIRRTL] Add Names to Side Effecting Statements, Make them Annotatable

seldridge, 2021-06-22:
Upstream FIRRTL added names to stop, printf, and verification statements (assert, assume, cover) in https://github.com/chipsalliance/firrtl/pull/2057.  

Support for annotating FIRRTL IR verification statements is being added to the Chisel in 3.5.0 (this is unreleased) in https://github.com/chipsalliance/chisel3/pull/1968.

Update the ODS representation of FIRRTL dialect to hold this name (which should be optional?). Update the annotation parsing logic to attach annotations to these side effecting statements.

ekiwi, 2021-06-22:
> Update the ODS representation of FIRRTL dialect to hold this name (which should be optional?)

I mostly made it optional for backwards compatibility reasons. Imho for the C++ firrtl compiler, it might make sense to just auto-generate unique names right after the parser in order to simplify the IR. For the Scala firrtl compiler we have to deal with IR nodes that don't come from the parser. The Scala firrtl compiler has `firrtl.transforms.EnsureNamedStatements` for transformations and backends that require named statements and I could see that becoming an "on by default" transform.

lattner, 2021-06-22:
If this is a string name, please don't make it an optional attribute, please just use "empty string" to mean missing.  This avoids the tristate problem where we have "present" "absent" and "empty" and have to rationalize that.

ekiwi, 2021-06-22:
> If this is a string name, please don't make it an optional attribute, please just use "empty string" to mean missing. This avoids the tristate problem where we have "present" "absent" and "empty" and have to rationalize that.

That is the implementation we chose for the Scala firrtl compiler. We just need to ensure that if a pass wants to use the name it needs to first verify that the String is non-empty. There are also now some special cases in our identifier name checking code that makes it so that more than one statement is allowed to be named `""`.

commit 202138bd (2021-07-09): [FIRRTL][SV] Add names/labels to side effect stmts

Add optional labels to the assert, assume, and cover SystemVerilog
operations.  Add optional labels to the assert, assume, cover, stop and
printf FIRRTL operations.

Flesh out lowering from these FIRRTL ops to their corresponding
SystemVerilog ops.  Emit the SystemVerilog ops as you'd expect for
assert, assume, and cover.  Names on stop and printf are dropped during
Verilog emission.

Labels/names are _optionally_ encoded using empty strings to represent
none.  This necessitates the proliferation of "custom<FooBar>"
directives in ODS.  This m [... cut]
PR #1398 (2021-07-09): [FIRRTL][SV] Add names/labels to side effect stmts
Add optional labels to the assert, assume, and cover SystemVerilog
operations.  Add optional labels to the assert, assume, cover, stop and
printf FIRRTL operations.

Flesh out lowering from these FIRRTL ops to their corresponding
SystemVerilog ops.  Emit the SystemVerilog ops as you'd expect for
assert, assume, and cover.  Names on stop and printf are dropped during
Verilog emission.

Labels/names are _optionally_ encoded using empty strings to represent
none.  This necessitates the proliferation of "custom<FooBar>"
directives in ODS.  This ma [... cut]
commit 52cc630f (2021-07-09): [FIRRTL][SV] Add names/labels to side effect stmts (#1398)

Add optional labels to the assert, assume, and cover SystemVerilog
operations.  Add optional labels to the assert, assume, cover, stop and
printf FIRRTL operations.

Flesh out lowering from these FIRRTL ops to their corresponding
SystemVerilog ops.  Emit the SystemVerilog ops as you'd expect for
assert, assume, and cover.  Names on stop and printf are dropped during
Verilog emission.

Labels/names are _optionally_ encoded using empty strings to represent
none.  This necessitates the proliferation of "custom<FooBar>"
directives in ODS. [... cut]
closed 2021-07-09 by seldridge