llvm/circt #1316: [Seq] Potential issue in python bindings test?

youngar, 2021-06-23:
While investigating an [integration failure](https://github.com/llvm/circt/runs/2894651160?check_suite_focus=true#step:11:158), I noticed the output of the test is indicating an error in the IR:
```
error: 'sv.reg' op should have non-empty name
```
This error is printed from [ExportVerilog](https://github.com/llvm/circt/blob/main/lib/Translation/ExportVerilog/ExportVerilog.cpp#L675). This seems like it's probably a problem that is being hidden by FileCheck.  (The `always_ff` error below is fixed in https://github.com/llvm/circt/pull/1315)

```
******************** TEST 'CIRCT :: Bindings/Python/dialects/seq.py' FAILED ********************
Script:
--
: 'RUN: at line 2';   /usr/bin/python3.8 /__w/circt/circt/integration_test/Bindings/Python/dialects/seq.py | /__w/circt/circt/llvm/install/bin/FileCheck /__w/circt/circt/integration_test/Bindings/Python/dialects/seq.py
--
Exit Code: 1

Command Output (stderr):
--
error: 'sv.reg' op should have non-empty name
error: 'sv.reg' op should have non-empty name
error: 'sv.reg' op should have non-empty name
/__w/circt/circt/integration_test/Bindings/Python/dialects/seq.py:75:11: error: CHECK: expected string not found in input
 # CHECK: always_ff @(posedge clk)
          ^
<stdin>:19:16: note: scanning from here
=== Verilog ===
               ^
<stdin>:40:2: note: possible intended match here
 always @(posedge clk)
 ^

Input file: <stdin>
Check file: /__w/circt/circt/integration_test/Bindings/Python/dialects/seq.py

-dump-input=help explains the following input dump.
```

youngar, 2021-06-23:
Besides the local issue in the test, a follow up question is: should ExportVerilog hard-fail when there is an error in the IR being exported?

mikeurbach, 2021-06-23:
bah, I will fix the actual issue in the test, thanks

teqdruid, 2021-06-29:
@mikeurbach Is this fixed? The integration tests pass for me.

mikeurbach, 2021-06-29:
I haven't gotten to it yet. I believe the test passes when it shouldn't :/

teqdruid, 2021-11-12:
I think I fixed this awhile back.

closed 2021-11-12 by teqdruid