llvm/circt #1324: [FIRRTL] Replace add/sub of negative literal with sub/add of positive literal

fabianschuiki, 2021-06-25:
Add the following canonicalizations:
```plain
add(x, -k) -> sub(x, k)
add(-k, x) -> sub(x, k)
sub(x, -k) -> add(x, k)
```

lattner, 2021-06-25:
I'd rather go the other way: please handle this as canonicalizing towards add.  Then have the verilog emitter do this peephole when printing.

The rationale here is that this makes other canonicalization patterns be able to only match add(x, cst) instead of having to look for both add(x,cst) and sub(x,cst).

fabianschuiki, 2021-06-28:
@lattner Good point. I've updated the issue description.

lattner, 2021-06-28:
Thanks!  In the description, it might be better expressed as `sub(x, k) -> add(x, -k)` since it applies even to positive constants

seldridge, 2021-07-13:
Note: this issue relates to SFC's `FixAddingNegativeLIteralsTransform`.  This is pre-Verilog emission cleanup to avoid a Verilator warning associated with code like `foo + -bar`.

E.g., the following:

```verilog
module Foo(
  input [7:0]  a,
  output [8:0] b
);
  assign b = $signed(a) + -8'sh2;
endmodule
```

Will produce a warning in Verilator:

```
%Warning-WIDTH: Foo.v:5:27: Operator NEGATE expects 9 bits on the LHS, but LHS's CONST '8'sh2' generates 8 bits.
                          : ... In instance Foo
    5 |   assign b = $signed(a) + -8'sh2;
      |                           ^
                ... Use "/* verilator lint_off WIDTH */" and lint_on around source to disable this message.
%Error: Exiting due to 1 warning(s)
```

Whereas canonicalizing to subtract will avoid it:

```verilog
module Foo(
  input [7:0]  a,
  output [8:0] b
);
  assign b = $signed(a) - 3'sh2;
endmodule
```

See the original SFC PR for more info: https://github.com/chipsalliance/firrtl/pull/1374.

seldridge, 2021-07-13:
Also, it looks like we're already doing the right thing for this specific example in CIRCT.  For the above circuit, CIRCT produces:

```verilog
module Foo(
  input  [7:0] x,
  output [8:0] z);

  assign z = {x[7], x} + 9'h1FE;
endmodule
```

lattner, 2021-07-13:
Ok, but there are two different things going on here:
1) We want canonicalization to produce a simple form that makes it easy to do pattern matching and rewriting.
2) We want to generate lint free output.

These are different things.  The subtract issue is #2, and should be handled in EmitVerilog.  I'll take a look at this.

lattner, 2021-07-13:
The printer side of this is handled here: 7cfbd471

lattner, 2021-07-13:
Canonicalization added here: 8277a623

closed 2021-07-13 by lattner