llvm/circt #1325: [FIRRTL] Replace add/sub with addw/subw if MSB is discarded

fabianschuiki, 2021-06-25:
In the case where the result of `add` or `sub` immediately throws away the expanded MSB, replace the operation with a non-expanding operator `addw` or `subw` respectively. This usually happens as in the following:
```plain
tail(add, 1) -> addw
tail(sub, 1) -> subw
bits(add, N, 0) -> addw
bits(sub, N, 0) -> subw
```

seldridge, 2021-07-13:
I'd suggest holding off on this unless we actually need it.  All of the `*w` operations (`addw`, `subw`, and `dshlw`) are non-standard FIRRTL operations that are only supposed to exist internal to the SFC.   They only exist as a bridge to map between FIRRTL semantics of operations (where widths are extending) to Verilog (where widths can be non-extending).  Furthermore, the intent of `addw` and `subw` is to remove an unnecessary tail when Verilog semantics doesn't require it.  Since CIRCT has Verilog dialect, I think this can be solved in a better way by canonicalization inside the Verilog dialect, e.g., removing unnecessary tails/bits if they are unnecessary given Verilog semantics.

For more info see the actual pass: [`ReplaceTruncatingArithmetic.scala`](https://github.com/chipsalliance/firrtl/blob/master/src/main/scala/firrtl/transforms/ReplaceTruncatingArithmetic.scala).

That said, the tests related to this would be useful to make sure we output the same thing: https://github.com/chipsalliance/firrtl/blob/master/src/test/scala/firrtlTests/ReplaceTruncatingArithmeticSpec.scala.

seldridge, 2021-07-13:
Here are the extracted challenge problems from the ReplaceTruncatingArithmetic spec.  

Ideally CIRCT should produce the same output Verilog for these cases.  I expect that a canonicalizer looking for a pattern like `extract(add(concat(0, x), concat(0, y)), ...) -> add(x, y)`.

Problems are below.  The FIRRTL input and HW MLIR is shown along with the _ideal_ Verilog to get out:

### 0
```scala
circuit Foo:
  module Foo:
    input x : UInt<8>
    input y : UInt<8>
    output z : UInt<8>
    z <= tail(add(x, y), 1)
```
```mlir
module attributes {firrtl.mainModule = "Foo"}  {
  hw.module @Foo(%x: i8, %y: i8) -> (%z: i8) {
    %false = hw.constant false
    %0 = comb.concat %false, %x : (i1, i8) -> i9
    %1 = comb.concat %false, %y : (i1, i8) -> i9
    %2 = comb.add %0, %1 : i9
    %3 = comb.extract %2 from 0 : (i9) -> i8
    hw.output %3 : i8
  }
}
```
```verilog
module Foo(
  input  [7:0] x,
  input  [7:0] y,
  output [7:0] z
);
  assign z = x + y;
endmodule
```

### 1 
```scala
circuit Foo:
  module Foo:
    input x : UInt<8>
    input y : UInt<8>
    output z : UInt<8>
    node n = add(x, y)
    z <= tail(n, 1)
```
```mlir
module attributes {firrtl.mainModule = "Foo"}  {
  hw.module @Foo(%x: i8, %y: i8) -> (%z: i8) {
    %false = hw.constant false
    %0 = comb.concat %false, %x : (i1, i8) -> i9
    %1 = comb.concat %false, %y : (i1, i8) -> i9
    %2 = comb.add %0, %1 : i9
    %3 = comb.extract %2 from 0 : (i9) -> i8
    hw.output %3 : i8
  }
}
```
```verilog
module Foo(
  input  [7:0] x,
  input  [7:0] y,
  output [7:0] z
);
  assign z = x - y;
endmodule
```

### 2 

```scala
circuit Foo:
  module Foo:
    input x : UInt<8>
    input y : UInt<8>
    output z : UInt<8>
    z <= tail(sub(x, y), 1)
```
```mlir
module attributes {firrtl.mainModule = "Foo"}  {
  hw.module @Foo(%x: i8, %y: i8) -> (%z: i8) {
    %false = hw.constant false
    %0 = comb.concat %false, %x : (i1, i8) -> i9
    %1 = comb.concat %false, %y : (i1, i8) -> i9
    %2 = comb.sub %0, %1 : i9
    %3 = comb.extract %2 from 0 : (i9) -> i8
    hw.output %3 : i8
  }
}
```
```verilog
module Foo(
  input  [7:0] x,
  input  [7:0] y,
  output [7:0] z
);
  assign z = x - y;
endmodule
```

### 3

```scala
circuit Foo:
  module Foo:
    input x : UInt<8>
    input y : UInt<8>
    output z : UInt<7>
    node n = sub(x, y)
    z <= tail(n, 2)
```
```mlir
module attributes {firrtl.mainModule = "Foo"}  {
  hw.module @Foo(%x: i8, %y: i8) -> (%z: i7) {
    %false = hw.constant false
    %0 = comb.concat %false, %x : (i1, i8) -> i9
    %1 = comb.concat %false, %y : (i1, i8) -> i9
    %2 = comb.sub %0, %1 : i9
    %3 = comb.extract %2 from 0 : (i9) -> i7
    hw.output %3 : i7
  }
}
```
```verilog
module Foo(
  input  [7:0] x,
  input  [7:0] y,
  output [6:0] z
);
  wire [8:0] n = x - y;
  assign z = n[6:0];
endmodule
```

lattner, 2021-07-13:
@fyquah FYI in case you're interested in this

fyquah, 2021-07-13:
Some care ought to be taken so we don't end up doing addition multiple times. For example:

```mlir
hw.module @Foo(%x: i8, %y: i8) -> (%z0: i8, %z1: i9) {
    %false = hw.constant false
    %0 = comb.concat %false, %x : (i1, i8) -> i9
    %1 = comb.concat %false, %y : (i1, i8) -> i9
    %2 = comb.add %0, %1 : i9
    %3 = comb.extract %2 from 0 : (i9) -> i8
    hw.output %3, %2 : (i8, i9)
  }
```

How about the following rewrite operations?

- `%addResult = add(x, y)` into `concat(0, add(extract(x) -> iN, extract(y) -> iN)))` where N is the widest usage of %addResult. if `N = width(%addResult)`, then don't rewrite.
- let the cannonicalizer for `extract(%addResult)` strip the leading zeros

fyquah, 2021-07-14:
A combination of https://github.com/llvm/circt/pull/1423 and https://github.com/llvm/circt/pull/1424 should fix this. Notably, this precise problem is taken care of in this test https://github.com/llvm/circt/pull/1424/files#diff-838d92f61a1cefe7d7b777fdd77339eef086cf28c27b61e568aefc344d189a53R356

fyquah, 2021-07-21:
https://github.com/llvm/circt/pull/1424 is now merged, which i believe a more general case of this.

closed 2021-07-21 by fyquah