llvm/circt #1343: Verifier does not catch mismatch in number of inputs

jenra-uwu, 2021-06-29:
Passing the following code through the verifier stage does not catch the mismatch in the number of inputs:
```
hw.module.extern @B(%d: i1, %e: i1) -> (%f: i1)

hw.module @AB(%w: i1, %x: i1, %i2: i2, %i3: i0) -> () {
  %w1, %y = hw.instance "b1" @B(%w) : (i1) -> (i1, i1)
}
```

This should error in the verification stage; instead, it is treated as valid and causes issues later on in the pipeline such as during Verilog code generation.

mikeurbach, 2021-06-29:
@teqdruid I know you have just been looking at this...

teqdruid, 2021-06-29:
I could have sworn we had that check, though I'm apparently wrong. It would explain why a few things I expected to fail haven't.

seldridge, 2024-02-09:
I believe this is fixed. The following (modern HW Dialect equivalent of the reported error):

``` mlir
module {
  hw.module.extern @B(in %d: i1, in %e: i1, out f: i1)
  hw.module @AB(in %w: i1, in %x: i1, in %i2: i2, in %i3: i0) {
    %w1, %y = hw.instance "b1" @B(d: %w: i1) -> (e: i1, f: i1)
  }
}
```

Produces the following errors:
```
# circt-opt 1343.mlir
1343.mlir:4:15: error: 'hw.instance' op has a wrong number of operands; expected 2 but got 1
    %w1, %y = hw.instance "b1" @B(d: %w: i1) -> (e: i1, f: i1)
              ^
1343.mlir:4:15: note: see current operation: %0:2 = "hw.instance"(%arg0) {argNames = ["d"], instanceName = "b1", moduleName = @B, parameters = [], resultNames = ["e", "f"]} : (i1) -> (i1, i1)
/Users/schuylereldridge/repos/github.com/seldridge/firrtl-snippets/circt/1343.mlir:2:3: note: module declared here
  hw.module.extern @B(in %d: i1, in %e: i1, out f: i1)
  ^
```

closed 2024-02-09 by seldridge