llvm/circt #1347: [FIRRTL] Remove CHIRRTL Behavioral Memory Constructs

seldridge, 2021-06-30:
FIRRTL Dialect currently includes everything that Chisel can emit.  This includes the _behavioral memory_ statements:

- `firrtl.cmem`
- `firrtl.smem`
- `firrtl.memoryport`

Add a pass that lowers these to `firrtl.memory` using the algorithms described in the following Scala FIRRTL Compiler (SFC) transforms:

- [`CInferMDir`](https://github.com/chipsalliance/firrtl/blob/master/src/main/scala/firrtl/passes/CInferMDir.scala)
- [`RemoveChirrtl`](https://github.com/chipsalliance/firrtl/blob/master/src/main/scala/firrtl/passes/RemoveCHIRRTL.scala)

Roughly, this needs to figure out whether each `firrtl.memoryport` is a read, write, or readwrite port and what the enable line is.

ekiwi, 2021-07-14:
If you are looking into how to handle "behavioral memories", are you going to try and address the issues with the current implementation?

- https://github.com/chipsalliance/firrtl/issues/727
- https://github.com/chipsalliance/firrtl/issues/1998
- - ...

youngar, 2021-12-10:
This has been implemented

closed 2021-12-10 by youngar