llvm/circt #1350: [FIRRTL] Verify Zero-width Behavior

seldridge, 2021-06-30:
It's unclear to me if CIRCT is aligning with Scala FIRRTL Compiler zero width behavior as defined in: 
- https://github.com/chipsalliance/firrtl/blob/master/src/test/scala/firrtlTests/ZeroWidthTests.scala

Verify that CIRCT is doing the right thing here, expand this issue with any differences that may exist, and close this once CIRCT passes all these tests.

- [x] Port
- [x] Add
- [ ] Mux: https://github.com/llvm/circt/pull/1349
- [ ] ???

lattner, 2021-07-02:
Nice, thank for tracking this.  Do you have a known ??? after mux?  If not, I'd recommend closing this issue when mux lands, then file new issues for any specific cases that come up, thanks!

seldridge, 2021-07-03:
> Do you have a known ??? after mux?

That was as far as I got with adding the tests from `ZeroWidthTests.scala`. 

> If not, I'd recommend closing this issue when mux lands, then file new issues for any specific cases that come up, thanks!

I'll just close this when #1349 goes in and open new issues as needed.

seldridge, 2021-07-03:
Closed the mux issue in #1349. Will re-open new issues as I pull failing tests out of the SFC for zero-width behavior.

closed 2021-07-03 by seldridge