llvm/circt #1356: Syntax error in generated System Verilog with struct in register.

mikeurbach, 2021-07-01:
I'm looking at a case like the following:

```mlir
hw.module @Test(%in: !hw.struct<a: i1, b: i1>, %clk: i1) -> (%out: !hw.struct<a: i1, b: i1>) {
  %0 = seq.compreg %in, %clk : !hw.struct<a: i1, b: i1>
  hw.output %0 : !hw.struct<a: i1, b: i1>
}
```

The desire is to register a packed struct. Technically, we're trying to register a packed array of packed structs, but just a struct is sufficient to demonstrate this issue.

I think it should be legal to register a packed struct, or a packed array of packed structs, but please stop me here if I'm wrong.

To lower this, I am using the following passes:

```sh
circt-opt test.mlir -lower-seq-to-sv -hw-legalize-names
```

Which produces:

```mlir
module  {
  hw.module @Test(%in: !hw.struct<a: i1, b: i1>, %clk: i1) -> (%out: !hw.struct<a: i1, b: i1>) {
    %_T = sv.reg  : !hw.inout<struct<a: i1, b: i1>>
    %0 = sv.read_inout %_T : !hw.inout<struct<a: i1, b: i1>>
    sv.alwaysff(posedge %clk)  {
      sv.passign %_T, %in : !hw.struct<a: i1, b: i1>
    }
    hw.output %0 : !hw.struct<a: i1, b: i1>
  }
}
```

Unless I'm missing something, so far so good. The issue is in the generated System Verilog.

I'm using:

```sh
circt-translate -export-verilog
```

Which produces:

```verilog
module Test(
  input  struct packed {logic a; logic b; } in,
  input                                     clk,
  output struct packed {logic a; logic b; } out);

  reg struct packed {logic a; logic b; } _T;	// <stdin>:3:11

  always @(posedge clk)	// <stdin>:5:5
    _T <= in;	// <stdin>:6:7
  assign out = _T;	// <stdin>:4:10, :8:5
endmodule
```

This seems like a faithful translation of the IR, but the issue is on the `reg` line.

Running that through Verilator with:

```sh
verilator -sv -lint-only
```

Produces these errors:

```sh
%Error: /tmp/a:6:7: syntax error, unexpected struct
    6 |   reg struct packed {logic a; logic b; } _T;
      |       ^~~~~~
%Error: /tmp/a:6:40: syntax error, unexpected '}'
    6 |   reg struct packed {logic a; logic b; } _T;
      |                                        ^
%Error: Exiting due to 2 error(s)
```

If I manually remove the `reg ` from the System Verilog, Verilator doesn't complain. @teqdruid has reported the same behavior from Questa. My reading of section 6.8 of the spec seems to confirm the removal of `reg ` is correct: in the grammar defined in section 6.8, `data_type` appears to support `reg` or a `struct` type, but not `reg struct ...`.

Is this something we can adjust in the ExportVerilog handling of registers of structs (and arrays of structs)? If there's consensus that this is the right line of thought, I can take that task.

mikeurbach, 2021-07-01:
Started poking around in ExportVerilog, and seeing where I could make the change I did manually. Not saying this is the best way to do it, but if I do this, Verilator doesn't complain: https://github.com/llvm/circt/pull/1361

fabianschuiki, 2021-07-01:
I think dropping `reg` is the right thing to do. In SV the `reg` type has become an alias for `logic`, so it's a type, not a keyword like `wire` that introduces a construct. `reg struct packed {...}` then essentially becomes the invalid `logic struct packed {...}`.

We could consider producing SystemVerilog-flavored output through a flag, and also replace `reg` with `logic` throughout, and `wire` with `logic` where applicable.

mikeurbach, 2021-07-01:
> replace reg with logic throughout, and wire with logic where applicable.

This is something @teqdruid mentioned as well.

@fabianschuiki do you mind taking a look at #1361 since you've chimed in here? I think it implements "dropping `reg`" in a decent way. I'm working on a little integration test to confirm that this actually works, but otherwise that PR should be ready to review.

mikeurbach, 2022-02-01:
This was closed in #1361.

closed 2022-02-01 by mikeurbach