llvm/circt #1360: unequal length in comparison operator

Ramlakshmi3733, 2021-07-01:
j < WIDTH
  A left 32-bit expression is compared to a right 4-bit expression.
  Comparing 'j' of type int
  with 'WIDTH' of type bit [3:0].

This becomes a VCS Error if the ULCO option is turned on.
There are other cases similar to this with flags such WMIA-L flag in VCS compilation, if the bit widths are not specified.
it expects the bits to be correctly balanced.
Have turned off these flags in the simulator and get a lint warning but have to turn them back on.

lattner, 2021-07-01:
Can you provide a small .fir file or .mlir file (either firrtl or hw dialect) that shows the issue here?

youngar, 2021-07-06:
Here is a test case.  The warning comes from the `RANDOM` initialization of the register.

```
; ./bin/
circuit Foo:
  module Foo:
    input c : Clock
    input i : UInt<1>
    output o : UInt<1>

    reg r : UInt<1>, c
    r <= UInt<1>("h0")

    when i :
      r <= UInt<1>("h1")
    o <= r
```

```verilog
// .. snip ...
module Foo(
  input  c, i,
  output o);

  reg r;	// ./reg.fir:7:5

  `ifndef SYNTHESIS	// ./reg.fir:7:5
    initial begin	// ./reg.fir:7:5
      `INIT_RANDOM_PROLOG_	// ./reg.fir:7:5
      `ifdef RANDOMIZE_REG_INIT	// ./reg.fir:7:5
        r = `RANDOM;	// ./reg.fir:7:5
      `endif
    end // initial
  `endif
  always @(posedge c)	// ./reg.fir:11:9
    r <= i;	// ./reg.fir:11:9
  assign o = r;	// ./reg.fir:2:10
endmodule
```

SFC Produces:
```verilog
module Foo(
  input   c,
  input   i,
  output  o
);
`ifdef RANDOMIZE_REG_INIT
  reg [31:0] _RAND_0;
`endif // RANDOMIZE_REG_INIT
  reg  r;
  assign o = r;
  always @(posedge c) begin
    r <= i; // @[]
  end
// ... snip ...
initial begin
  `ifdef RANDOMIZE
    `ifdef INIT_RANDOM
      `INIT_RANDOM
    `endif
    `ifndef VERILATOR
      `ifdef RANDOMIZE_DELAY
        #`RANDOMIZE_DELAY begin end
      `else
        #0.002 begin end
      `endif
    `endif
`ifdef RANDOMIZE_REG_INIT
  _RAND_0 = {1{`RANDOM}};
  r = _RAND_0[0:0];
`endif // RANDOMIZE_REG_INIT
  `endif // RANDOMIZE
end // initial
`ifdef FIRRTL_AFTER_INITIAL
`FIRRTL_AFTER_INITIAL
`endif
`endif // SYNTHESIS
endmodule
```

lattner, 2021-07-06:
awesome, thanks

youngar, 2021-07-07:
Using the following FIRRTL and annotations json files:
```
circuit Foo:
  module Foo:
    inst bar of Bar

  extmodule Bar:
    defname = RealBar
    parameter WIDTH = 68
```
```json
[{
    "class": "firrtl.transforms.BlackBoxInlineAnno",
    "target":"Foo.Bar",
    "name": "Bar.sv",
    "text":"module RealBar #(parameter WIDTH=32) ();\n int i;\n initial begin\n  if(i<WIDTH)\n   $display(\"hello\");\n end\nendmodule"
}]
```
When compiling it with `firtool -verilog ./test.fir -o test.sv --annotation-file=test.json` produces:

```verilog
// Standard header to adapt well known macros to our needs.

module Foo();
  RealBar #(
    .WIDTH(8'd68)
  ) bar ();	// ./test.fir:3:5
endmodule

// external module RealBar

// ----- 8< ----- FILE "./bar.v" ----- 8< -----

module RealBar #(parameter WIDTH=32) ();
 int i;
 initial begin
  if(i<WIDTH)
   $display("hello");
 end
endmodule	// ./test.fir:5:13

// ----- 8< ----- FILE "firrtl_black_box_resource_files.f" ----- 8< -----

./bar.v	// ./test.fir:1:1
```

After deleting the file list (last line), you can compile it with VCS using 
```
vcs -full64 -notice -error=BMII,CFCOR,LFCOR,VCM-HFUFR,ULCO,WMIA-L -timescale=1ns/10ps -reportstats +systemverilogext+sv +rad test.sv
```

Produces the following error:
```
Error-[ULCO] Unequal length in comparison operator
test.sv, 17
RealBar, "(i < WIDTH)"
  A left 32-bit expression is compared to a right 8-bit expression.
  Comparing 'i' of type int
  with 'WIDTH' of type bit [7:0].

1 error
```

This is due to the the way the WIDTH parameter is passed: ` .WIDTH(8'd68)`.

For reference, running SFC with `firrtl-1.5-SNAPSHOT -i test.fir -o test -faf test.json` produces:

```verilog
module Foo(
);
  RealBar #(.WIDTH(68)) bar (
  );
endmodule
```

The way we pass parameters was changed in https://github.com/llvm/circt/commit/cfd641a71ee721e78234518db2a6f29c3648654b

lattner, 2021-07-11:
Fixed in 9f15227d, thanks.

youngar, 2021-07-12:
I think this problem is fixed, but I want to point out that we are still differing from SFC in how we pass parameters `>32` bits and negative values.  `>32` bit parameters are passed with an explicit width,  and printing negative numbers are subject to the width of the APInt (e.g. `1'b1` and  `4'b1111` are both `-1`).  I have no idea if this is a problem or not!

```
circuit Foo: 
  module Foo:
    inst bar of Bar
  extmodule Bar:
    defname = RealBar
    parameter WIDTH0 = 0
    parameter WIDTH1 = 4
    parameter WIDTH2 = 6812312123
    parameter WIDTH3 = -1
    parameter WIDTH4 = -88888888888888888
    parameter Wtricky = 4294967295
```

SFC: 
```verilog
module Foo(
);
  RealBar
    #(.WIDTH0(0), .WIDTH1(4), .WIDTH2(33'd6812312123), .WIDTH3(-1), .WIDTH4(-58'sd88888888888888888), .Wtricky(32'd4294967295))
    bar (
  );
endmodule
```

MFC:
```verilog
module Foo();
  RealBar #(
    .WIDTH0(0),
    .WIDTH1(4),
    .WIDTH2(6812312123),
    .WIDTH3(15),
    .WIDTH4(295059016290463936968),
    .Wtricky(4294967295)
  ) bar ();	// ./ram.fir:3:5
endmodule

// external module RealBar
```

teqdruid, 2021-07-12:
This broke the output for >32 bit values in Verilator:

```
----- STDERR ------
%Error: /__w/circt/circt/build/integration_test/ESI/cosim/Output/basic.mlir.tmp1.sv:81:19: Too many digits for 32 bit number: 10578209918096690139
        ... As that number was unsized ('d...) it is limited to 32 bits (IEEE 1800-2017 5.7.1)
        ... Suggest adding a size to it.
   81 |     .RECV_TYPE_ID(10578209918096690139),
      |                   ^~~~~~~~~~~~~~~~~~~~
%Error: /__w/circt/circt/build/integration_test/ESI/cosim/Output/basic.mlir.tmp1.sv:83:19: Too many digits for 32 bit number: 10578209918096690139
   83 |     .SEND_TYPE_ID(10578209918096690139),
      |                   ^~~~~~~~~~~~~~~~~~~~
%Error: /__w/circt/circt/build/integration_test/ESI/cosim/Output/basic.mlir.tmp1.sv:193:19: Too many digits for 32 bit number: 14333817230841524124
  193 |     .RECV_TYPE_ID(14333817230841524124),
      |                   ^~~~~~~~~~~~~~~~~~~~
%Error: /__w/circt/circt/build/integration_test/ESI/cosim/Output/basic.mlir.tmp1.sv:195:19: Too many digits for 32 bit number: 9939105206106407124
  195 |     .SEND_TYPE_ID(9939105206106407124),
      |                   ^~~~~~~~~~~~~~~~~~~
%Error: /__w/circt/circt/build/integration_test/ESI/cosim/Output/basic.mlir.tmp1.sv:312:19: Too many digits for 32 bit number: 15822124641382404136
  312 |     .RECV_TYPE_ID(15822124641382404136),
      |                   ^~~~~~~~~~~~~~~~~~~~
%Error: /__w/circt/circt/build/integration_test/ESI/cosim/Output/basic.mlir.tmp1.sv:314:19: Too many digits for 32 bit number: 15822124641382404136
  314 |     .SEND_TYPE_ID(15822124641382404136),
      |                   ^~~~~~~~~~~~~~~~~~~~
%Error: /__w/circt/circt/build/integration_test/ESI/cosim/Output/basic.mlir.tmp1.sv:340:19: Too many digits for 32 bit number: 14745270011869700302
  340 |     .RECV_TYPE_ID(14745270011869700302),
      |                   ^~~~~~~~~~~~~~~~~~~~
%Error: /__w/circt/circt/build/integration_test/ESI/cosim/Output/basic.mlir.tmp1.sv:342:19: Too many digits for 32 bit number: 17519082812652290511
  342 |     .SEND_TYPE_ID(17519082812652290511),
      |                   ^~~~~~~~~~~~~~~~~~~~
```

lattner, 2021-07-13:
Thank you for the testcase andrew, I'll take a look!

lattner, 2021-07-13:
Thanks again, fixed in 20819ceb

closed 2021-07-13 by lattner