llvm/circt #1368: remove EmitVerilog/sv-structs.mlir integration test

lattner, 2021-07-03:
Rationale: Integration tests are expensive to maintain and run.  Struct emission is already handled by a unit test.  We need integration tests for complicated end to end things, not peephole issues in the verilog printer.  I think we should remove this.

Any concerns?

lattner, 2021-07-03:
@mikeurbach

mikeurbach, 2021-07-05:
Sorry, I commented on the original PR but I guess we should keep the discussion here. I think @teqdruid just wanted to confirm that we can actually put packed structs into registers, and the tools do the right thing. I'm happy to have confirmed this locally without requiring a CIRCT integration test covering this behavior. And of course we have the unit test that ensures ExportVerilog emits struct declarations according to the syntax I've confirmed. I can remove the integration test. Does that sound ok @teqdruid?

mikeurbach, 2021-07-05:
Also, while doing this cleanup, I see we have a `sv-interfaces.mlir`, which seems conceptually similar to `sv-structs.mlir`. It's an integration test that confirms the constructs we are emitting (and have unit tests covering) do indeed work as expected. Seems to beg the question if we should just delete `sv-interfaces.mlir` as well.

closed 2021-07-05 by mikeurbach

lattner, 2021-07-05:
Thanks!  yes, I understand the goal.  Such a test is a good thing to do for local development to verify that the fix works as intended, but the checked in unit test should cover to make sure we generate the (manually verified) correct logic.  Thank you for taking care of this!

teqdruid, 2021-07-05:
The other reason to have these "does this feature work as we want it to" tests is to verify that all the simulators/synthesizers behave the way we intend. We should really have a set of "compatibility" tests for these sort of tests.

lattner, 2021-07-06:
Sure that can make sense, but I'd recommend that be a "different thing" than the unit tests - a dedicated testsuite.  LLVM has the analogous thing for C/C++ programs in its [test-suite](https://llvm.org/docs/TestSuiteGuide.html) repository, which is the ultimate integration test in some ways.  When bugs are found by it, they get distilled into unit tests and checked into the compiler's testsuite.

teqdruid, 2021-07-06:
Yeah, that's what I was talking about by a set of "compatibility" tests.