llvm/circt #1373: More MLIRSwift comments on attributes/types + registration

lattner, 2021-07-06:
Hey George,

I'm getting back to being able to do things, here's another pass over type/attribute registration stuff. 
 It is looking really really nice.  Let me know if you agree with something, and I'll be happy to help with changes.  Some minor things (everything I say about types applies to attrs as well):

- TypeRepresentable/AttributeRepresentable implements `==` but isn't Equatable.  Should this be added?
- I think we can drop this comment in Type.swift, I don't think we want people traffic'ing in MlirType outside of MLIRSwift itself, WDYT?:

```
- note: We originally attempted just conforming `MlirType` to `TypeRepresentable`, but this conformance did not propagate to dependent packages. This seems to be due to some issue with ODR and the clang importer (since dependent packages often have their own targets which import the MLIR headers).
```

- The `MlirTypeID` imported type is poorly named on the C++ side of things, it should be be something like `MlirTypeOrAttributeID`.

 - Why do you encourage scoped resolution construction syntax instead of using a normal init, e.g.  `.index(in: context)` instead of `IndexType(in: context)`?  This leads to redundant API in some cases (e.g. for FunctionType which has both forms), seems to unnecessarily clutter the API of TypeProtocol, increases the odds of name collisions on `TypeProtocol` (conflicting type names can be resolved with module scope resolution like `FIRRTL.IntType` or whatever, but the method name cannot), and is unnecessarily unusual from a Swift API design perspective.  I'd recommend dropping the `.index(in: ctx)` form and adding `IndexType(in: context)`.

 - The `TypeProtocol.wrappingInMemRef` pattern is more defensible, but should it be named `wrappedInMemRef` or is wrapping the right name?

 - I'd recommend following the MLIR convention for attribute naming: keep the name `Attribute` as you're using it, but rename `DictionaryAttribute` to `DictionaryAttr`.  This aligns with MLIR conventions better, and there is no need for the extra verbosity.

The name MLIRSwift.Binding is extremely generic and doesn't convey much about what this thing is.  The function of this is to provide the "MlirTypeId -> Swift Pretty Attribute" and the same for Types.  It also looks like it has an inheritance thing going on (where there can be a tower of these things).  I don't think this will actually work though: all the "bindings" need to be registered before the IR is interacted with (otherwise some "old" things will get mapped to OpaqueAttribute and "new" things will get mapped to "YourSpecificAttribute" and Swift type conversions will fail.

I'd recommend that we have a single global registry associated with the context, eliminating Binding as a public type: Context is enough for all the global things.  It is still useful to have this split off as its own internal thing as an implementation detail of Context, but I'd rename it something like `TypeAndAttributeRegistry` to be more specific.

In addition to the bigger design issue (where we can't actually support scoped and transient binding sets with MLIRContext), this addresses a few other minor issues:

 1) Passing around Binding as designed isn't very cheap because it contains some COW types so it will induce unnecessary ARC traffic.
 2) Having this as part of Context will eliminate the need for the simple/complex builder thing: everyone always has/needs the context anyway.
 3) This moves type/attribute registration into a more explicit registration hook that is called when the dialect is registered, instead of the implicitly-lazily-initialized global pattern that is currently going on.

-Chris

lattner, 2021-07-06:
Oops, wrong repo plz disregard!

closed 2021-07-06 by lattner

teqdruid, 2021-07-06:
I was wondering what this was in response to!

> * The `MlirTypeID` imported type is poorly named on the C++ side of things, it should be be something like `MlirTypeOrAttributeID`.

It's actually more generic than that as it is also used for traits and opinterfaces (and maybe other things).

lattner, 2021-07-06:
Yeah you're right, I filed this bugzilla to track this: https://bugs.llvm.org/show_bug.cgi?id=50997