llvm/circt #1379: Clang Format CI test broken

youngar, 2021-07-06:
In our pull request CI we test that clang-format does not make any changes.  This is currently not working because the command to find the base commit before the PR commits is broken:

```
Run if [[ ! "$DIFF_COMMIT_NAME" == *"HEAD"* ]]; then
  if [[ ! "$DIFF_COMMIT_NAME" == *"HEAD"* ]]; then
    git fetch --recurse-submodules=no origin $DIFF_COMMIT_NAME
    DIFF_COMMIT_SHA=$( git rev-parse origin/$DIFF_COMMIT_NAME )
  else
    DIFF_COMMIT_SHA=$( git rev-parse $DIFF_COMMIT_NAME )
  fi
  echo "DIFF_COMMIT=$DIFF_COMMIT_SHA" >> $GITHUB_ENV
  shell: sh -e {0}
  env:
    DIFF_COMMIT_NAME: main
/__w/_temp/146db661-d59d-4bcf-8d76-20be01aaa8fe.sh: 1: [[: not found
fatal: ambiguous argument 'main': unknown revision or path not in the working tree.
Use '--' to separate paths from revisions, like this:
'git <command> [<revision>...] -- [<file>...]'
Error: Process completed with exit code 128.
```

Example: https://github.com/llvm/circt/pull/1304/checks?check_run_id=2961500533#step:12:16

teqdruid, 2021-07-06:
*sigh* This probably broke months ago when we switched to Docker.

mikeurbach, 2021-07-06:
Damn, didn't notice that. I did notice this for Python, and the fix is simple: https://github.com/llvm/circt/pull/1193. This has been an issue since we switched to using Docker.

youngar, 2021-07-06:
Thanks for pointing out the fix! Looks easy so I'll open a PR

youngar, 2021-07-06:
The workflow rules are all marked `continue-on-error`, which I'm guessing is why this went unnoticed for so long.  Should I get rid of that?

teqdruid, 2021-07-06:
No. That's to allow the upload diff step to run when the format step finds an issue. Probably not the best way, so feel free to re-work if you think it's necessary.

youngar, 2021-07-07:
Fixed in: https://github.com/llvm/circt/pull/1380

closed 2021-07-07 by youngar