llvm/circt #1381: [FIRRTL] width inference (reg) false positive

drom, 2021-07-07:
The following FIRRTL program

```python
circuit top_mod:
  module top_mod:
    input clk: Clock
    input d: UInt
    output q: UInt
    reg _q: UInt, clk
    _q <= d
    q <= _q
```

Compiled with `firtool` produces this Verilog:

```verilog
module top_mod(
  input clk);

endmodule
```

Compiled with `firrtl-1.5-SNAPSHOT` produces this Error:

```
Exception in thread "main" firrtl.passes.PassExceptions: 
firrtl.passes.CheckWidths$UninferredWidth:  : Uninferred width for target below.serialize}. (Did you forget to assign to it?)
    circuit top_mod:
    └── module top_mod:
        └── d
firrtl.passes.CheckWidths$UninferredWidth:  : Uninferred width for target below.serialize}. (Did you forget to assign to it?)
    circuit top_mod:
    └── module top_mod:
        └── q
firrtl.passes.CheckWidths$UninferredWidth:  : Uninferred width for target below.serialize}. (Did you forget to assign to it?)
    circuit top_mod:
    └── module top_mod:
        └── _q
firrtl.passes.PassException: 3 errors detected!
```

seldridge, 2021-07-07:
Nice find!

It looks like this can be simplified even further to:

```mlir
firrtl.module @top_mod(in %a: !firrtl.uint, out %b: !firrtl.uint) {
  firrtl.connect %b, %a : !firrtl.uint, !firrtl.uint
}
```

Infer widths infers both `%a` and `%b` as being zero-width. After infer widths, this is:

```mlir
// -----// IR Dump After InferWidths //----- //
firrtl.circuit "top_mod"   {
  firrtl.module @top_mod(in %a: !firrtl.uint<0>, out %b: !firrtl.uint<0>) {
    firrtl.connect %b, %a : !firrtl.uint<0>, !firrtl.uint<0>
  }
}
```

fabianschuiki, 2021-07-07:
Awesome find! I hope the merging of CheckWidths into InferWidths will fix this. Right now it might be that InferWidths fills in a 0 width in some cases, which then doesn't get caught in CheckWidths. Thanks @drom :+1:

commit 0fd23468 (2021-07-14): [FIRRTL][InferWidths] Fix unconstrained widths inferred to zero

Fix an issue in `InferWidths` where an unconstrained width could be
inferred to zero. Curiously, this only happens if the unconstrained
width is used as part of another constraint expression, e.g. a connect
of one unconstrained width to another. This fix makes the width
calculation properly return a `None` value instead of `0` for the width.
Also fixes a nasty subtle bug where the cycle breaking code was not
properly balanced.

Fixes #1381.
closed 2021-07-14 by fabianschuiki