llvm/circt #1385: [FIRRTL] Width mismatch in random register initialization

youngar, 2021-07-07:
Register initialization does not take into account the size of the register.

Using a 3-bit register and compiling the resultant verilog with with `vcs -full64 -error\=BMII,CFCOR,LFCOR,VCM-HFUFR,ULCO,WMIA-L +systemverilogext+sv +rad +define+RANDOMIZE_REG_INIT reg.sv` produces:

```
Error-[WMIA-L] Width mismatch in assignment
reg.sv, 50
  Width mismatch between LHS and RHS is found in assignment:
  The following 32-bit wide expression is assigned to a 3-bit LHS target:
  Source info: r = $random;
  Expression: r
```

For example, for a 3 bit register, MFC produces:
```verilog
module Foo(
  input        c, i,
  output [2:0] o);

  reg [2:0] r;	// ./reg.fir:7:5
   // ... snip...
  `ifndef SYNTHESIS	// ./reg.fir:7:5
    initial begin	// ./reg.fir:7:5
      `INIT_RANDOM_PROLOG_	// ./reg.fir:7:5
      `ifdef RANDOMIZE_REG_INIT	// ./reg.fir:7:5
        if (~i)	// ./reg.fir:7:5
          r = `RANDOM;	// ./reg.fir:7:5
      `endif
    end // initial
  `endif
  assign o = r;	// ./reg.fir:2:10
endmodule
```

While SFC produces:
```verilog
module Foo(
  input        c,
  input        i,
  output [2:0] o
);
`ifdef RANDOMIZE_REG_INIT
  reg [31:0] _RAND_0;
`endif // RANDOMIZE_REG_INIT
  reg [2:0] r;
   // ... snip...
initial begin
  `ifdef RANDOMIZE
    `ifdef INIT_RANDOM
      `INIT_RANDOM
    `endif
    `ifndef VERILATOR
      `ifdef RANDOMIZE_DELAY
        #`RANDOMIZE_DELAY begin end
      `else
        #0.002 begin end
      `endif
    `endif
`ifdef RANDOMIZE_REG_INIT
  _RAND_0 = {1{`RANDOM}};
  r = _RAND_0[2:0];
`endif // RANDOMIZE_REG_INIT
  `endif // RANDOMIZE
end // initial
`ifdef FIRRTL_AFTER_INITIAL
`FIRRTL_AFTER_INITIAL
`endif
`endif // SYNTHESIS
endmodule
```

This FIRRTL file can be used to reproduce the issue.  Its is a little long to keep SFC from optimizing away the register.
```
circuit Foo:
  module Foo:
    input c : Clock
    input i : UInt<1>
    output o : UInt<3>

    reg r : UInt<3>, c
    r <= UInt<1>("h0")

    when i :
      r <= UInt<3>("h1")

    o <= r
```

fabianschuiki, 2021-07-07:
Yeah `$random` is nasty in assuming 32 bit output. Maybe we could switch to `std::randomize()`, which takes reference to a variable to randomize (see "18.12 Randomization of scope variables—std::randomize()" in the standard).

lattner, 2021-07-07:
Wow that's crazy, I had no idea SV had that.  Is it widely supported by all the tools in the ecosystem?  How does SFC handle this?

lattner, 2021-07-07:
Oh I see what is going on here.  I think a straight-forward way to handle this is to model the "`RANDOM" expression as an i32 expression in the IR always.  Using it as an i3 would require a truncation, and the verilog emitter will do the right thing.

youngar, 2021-07-07:
We would need to handle extending to `>32` as well, here is a 42 bit integer:
```verilog
reg [63:0] _RAND_0;
reg [41:0] r;
_RAND_0 = {2{`RANDOM}};
r = _RAND_0[41:0];
```

lattner, 2021-07-07:
Right, I'm suggesting that hte LowerToHW pass handle this.  In this case, it would just use two different i32 "`RANDOM`" expressions, concat them, then truncate down to 42-bit.  It should all compose right.

fabianschuiki, 2021-07-07:
> @lattner Is it widely supported by all the tools in the ecosystem?

I have a bad feeling that it isn't...

I like the approach of requiring `RANDOM` to be 32 bits and then truncating/concatenating as necessary. Less "standard surface" we expose to tools, the less disagreement on standard implementation we get.

seldridge, 2021-07-07:
I have a very old macro I used to use for this: https://github.com/seldridge/verilog/blob/master/include/verilog_math.vh#L70.

lattner, 2021-07-07:
I have a few cycles this afternoon.  Unless someone is already working on this, I can take it.

lattner, 2021-07-08:
First half of this is fixed in 490d2aa8, but there is another issue that needs to be resolved.

lattner, 2021-07-11:
Side effect issue is fixed in 72f4ec62

closed 2021-07-11 by lattner