llvm/circt #1393: Roadmap for the static scheduling infrastructure

jopperm, 2021-07-08:
### Modeling of problems
- [ ] **Basic**: Operators have integer latency. Solution: Integer start times for all operations.
   #1268
   #1392 
- [ ] **Cyclic** extends **Basic**: Dependences may have distance. Solution: Initiation interval (II).
- [ ] ...

### Algorithms
- [x] List scheduler for the **Basic** problem.
   #1358 
- [ ] ...

### Utilities
- [ ] Cycle detector/enumerator
- [ ] Modulo reservation table
- [ ] ...

### External tools
- [ ] (I)LP solver
- [ ] ...

*Let me know if you are interested in a specific problem/algorithm, I'm happy to reprioritize!*

mikeurbach, 2021-07-08:
Regarding cycle detection, I saw @hanchenye recently started an analysis to detect combinational cycles in the FIRRTL dialect: https://github.com/llvm/circt/pull/1388. There is also a pass in the Handshake dialect that we've both had our hands on, which has the ability to look for cycles in the dataflow in order to insert buffers: https://github.com/llvm/circt/blob/037ab746e4e639df96f6eae4b8342d776122e00e/lib/Conversion/StandardToHandshake/StandardToHandshake.cpp#L1511.

I wonder if we can standardize these into a generic cycle detection utility. I'm wondering if it can be done generically on opaque `Operation *`s... If not, it seems like a good candidate for some OpInterface(s) somewhere, to guide such a utility.

hanchenye, 2021-07-08:
@mikeurbach There're some infrastructures in LLVM, [GraphTraits](https://llvm.org/doxygen/GraphTraits_8h_source.html) and [SCCIterator](https://llvm.org/doxygen/SCCIterator_8h_source.html), for creating a view of graph on top of the IR and iterating on SCCs. This should work well on graph regions, such as handshake function.

jopperm, 2021-07-08:
Oh the SCC iterators look very promising, should be straightforward to define the graph traits on top of scheduling problem. Thanks!

mikeurbach, 2021-07-08:
Interesting, that's good to know!

jopperm, 2022-03-13:
Retiring this generic tracker because most of the initial vision for the infra has been fleshed out. Future projects (problems, algorithms, integrations, ...) should just use their own issues/PRs/... and the [Scheduling] label.

closed 2022-03-13 by jopperm