llvm/circt #1415: [ExportVerilog] Wrong bitwidth being printed on module parameters

teqdruid, 2021-07-13:
Given the following IR:

```
hw.module.extern @Cosim_Endpoint(%clk: i1, %rstn: i1, %DataOutReady: i1, %DataInValid: i1, %DataIn: none) -> (%DataOutValid: i1, %DataOut: none, %DataInReady: i1)
hw.module @ints(%clk: i1, %rstn: i1) {
  %DataOutValid, %DataOut, %DataInReady = hw.instance "TestEP" @Cosim_Endpoint(%clk, %rstn, %DataInReady, %DataOutValid, %DataOut) {parameters = {ENDPOINT_ID = 1 : i32, RECV_TYPE_ID = 0578209918096690139 : ui64, RECV_TYPE_SIZE_BITS = 128 : i32, SEND_TYPE_ID = 3057809918096690139 : ui64, SEND_TYPE_SIZE_BITS = 128 : i32}} : (i1, i1, i1, i1, !hw.array<128xi1>) -> (i1, !hw.array<128xi1>, i1)
}
```

ExportVerilog currently produces:
```
// external module Cosim_Endpoint

module ints(
  input clk, rstn);

  wire         TestEP_DataOutValid;     // basic.mlir:3:43
  wire [127:0] TestEP_DataOut;  // basic.mlir:3:43
  wire         TestEP_DataInReady;      // basic.mlir:3:43

  Cosim_Endpoint #(
    .ENDPOINT_ID(1),
    .RECV_TYPE_ID(60'd578209918096690139),
    .RECV_TYPE_SIZE_BITS(128),
    .SEND_TYPE_ID(62'd3057809918096690139),
    .SEND_TYPE_SIZE_BITS(128)
  ) TestEP (    // basic.mlir:3:43
    .clk          (clk),
    .rstn         (rstn),
    .DataOutReady (TestEP_DataInReady), // basic.mlir:3:43
    .DataInValid  (TestEP_DataOutValid),        // basic.mlir:3:43
    .DataIn       (TestEP_DataOut),     // basic.mlir:3:43
    .DataOutValid (TestEP_DataOutValid),
    .DataOut      (TestEP_DataOut),
    .DataInReady  (TestEP_DataInReady)
  );
endmodule
```

The problematic lines being:
```
    .RECV_TYPE_ID(60'd578209918096690139),
    .SEND_TYPE_ID(62'd3057809918096690139),
```

Since the module being [instantiated explicitly requires a 64 bit value](https://github.com/llvm/circt/blob/8277a62325658b4a17b14f83cb2caa09a78b9ee2/include/circt/Dialect/ESI/cosim/Cosim_Endpoint.sv#L20), Verilator (and presumably other simulators) error out since the widths don't match.

ExportVerilog used to and needs to produce:
```
    .RECV_TYPE_ID(64'd578209918096690139),
    .SEND_TYPE_ID(64'd3057809918096690139),
```

We should go back to using the type information in the IntegerAttribute (width and signedness) since sometimes Verilog requires that it match rather than relying on implicit metadata from APInt. See https://github.com/llvm/circt/commit/cfd641a71ee721e78234518db2a6f29c3648654b.

Is there a specific reason that we stopped doing this? I understand the part about not printing the width for values under 32 bits to better accommodate external modules with unknown/unmodeled parameter types.

teqdruid, 2021-07-14:
Also, the sign being printed is wrong. This snippet is supposed to (and used to) print a non-negative number, but now it doesn't:

```
hw.module.extern @ModuleWithStronglyTypedParameters() -> ()
hw.module @test() {
  hw.instance "TestEP" @ModuleWithStronglyTypedParameters() {parameters = { RECV_TYPE_ID = 10578209918096690139}} : () -> ()
}
```

lattner, 2021-07-25:
My understanding is that this is all fixed up now, sorry again for the breakage!

closed 2021-07-25 by lattner

teqdruid, 2021-07-26:
Yes it is! Sorry I forgot to close this ticket.