llvm/circt #1416: [HW][FIRRTL] Better Bits Inlining

seldridge, 2021-07-13:
The Scala FIRRTL Compiler has a dedicated transform that inlines bit extraction for prettier Verilog.  We should do something similar with how Verilog is exported.  (I am not advocating for a pass that does this.)  It looks like either CSE should be relaxed here or `ExportVerilog` should be inlining bit extracts during emission.

This is purely a stylistic thing.  @drom: do you have any thoughts on what is a good strategy here?

The following is taken directly from the SFC's [`VerilogEmitterTests.scala`](https://github.com/chipsalliance/firrtl/blob/87ab555023760e7fe6f517c5776975bbc93ebe8c/src/test/scala/firrtlTests/VerilogEmitterTests.scala#L91):

```scala
circuit InlineBits :
  module InlineBits :
    input a: UInt<4>
    output b: UInt<1>
    output c: UInt<3>
    output d: UInt<2>
    output e: UInt<2>
    output f: UInt<2>
    output g: UInt<2>
    output h: UInt<2>
    output i: UInt<2>
    output j: UInt<2>
    output k: UInt<1>
    output l: UInt<1>
    output m: UInt<1>
    output n: UInt<1>
    output o: UInt<2>
    output p: UInt<2>
    output q: UInt<2>
    output r: UInt<1>
    output s: UInt<2>
    output t: UInt<2>
    output u: UInt<1>
    b <= bits(a, 2, 2)
    c <= bits(a, 3, 1)
    d <= head(a, 2)
    e <= tail(a, 2)
    f <= bits(bits(a, 3, 1), 2, 1)
    g <= bits(head(a, 3), 1, 0)
    h <= bits(tail(a, 1), 1, 0)
    i <= bits(shr(a, 1), 1, 0)
    j <= head(bits(a, 3, 1), 2)
    k <= head(head(a, 3), 1)
    l <= head(tail(a, 1), 1)
    m <= head(shr(a, 1), 1)
    n <= tail(bits(a, 3, 1), 2)
    o <= tail(head(a, 3), 1)
    p <= tail(tail(a, 1), 1)
    q <= tail(shr(a, 1), 1)
    r <= shr(bits(a, 1, 0), 1)
    s <= shr(head(a, 3), 1)
    t <= shr(tail(a, 1), 1)
    u <= shr(shr(a, 1), 2)
```

In MLIR (with optimizations):

```mlir
module attributes {firrtl.mainModule = "InlineBits"}  {
  hw.module @InlineBits(%a: i4) -> (%b: i1, %c: i3, %d: i2, %e: i2, %f: i2, %g: i2, %h: i2, %i: i2, %j: i2, %k: i1, %l: i1, %m: i1, %n: i1, %o: i2, %p: i2, %q: i2, %r: i1, %s: i2, %t: i2, %u: i1) {
    %0 = comb.extract %a from 2 : (i4) -> i1
    %1 = comb.extract %a from 1 : (i4) -> i3
    %2 = comb.extract %a from 2 : (i4) -> i2
    %3 = comb.extract %a from 0 : (i4) -> i2
    %4 = comb.extract %a from 1 : (i4) -> i2
    %5 = comb.extract %a from 3 : (i4) -> i1
    %6 = comb.extract %a from 1 : (i4) -> i1
    hw.output %0, %1, %2, %3, %2, %4, %3, %4, %2, %5, %0, %5, %6, %4, %3, %4, %6, %2, %4, %5 : i1, i3, i2, i2, i2, i2, i2, i2, i2, i1, i1, i1, i1, i2, i2, i2, i1, i2, i2, i1
  }
}
```

CIRCT produces the following Verilog:
```verilog
module InlineBits(
  input  [3:0] a,
  output       b,
  output [2:0] c,
  output [1:0] d, e, f, g, h, i, j,
  output       k, l, m, n,
  output [1:0] o, p, q,
  output       r,
  output [1:0] s, t,
  output       u);

  wire _T = a[2];	// InlineBits.fir:24:10
  wire [1:0] _T_0 = a[3:2];	// InlineBits.fir:26:10
  wire [1:0] _T_1 = a[1:0];	// InlineBits.fir:27:10
  wire [1:0] _T_2 = a[2:1];	// InlineBits.fir:29:10
  wire _T_3 = a[3];	// InlineBits.fir:33:10
  wire _T_4 = a[1];	// InlineBits.fir:36:10
  assign b = _T;	// InlineBits.fir:2:10
  assign c = a[3:1];	// InlineBits.fir:2:10, :25:10
  assign d = _T_0;	// InlineBits.fir:2:10
  assign e = _T_1;	// InlineBits.fir:2:10
  assign f = _T_0;	// InlineBits.fir:2:10
  assign g = _T_2;	// InlineBits.fir:2:10
  assign h = _T_1;	// InlineBits.fir:2:10
  assign i = _T_2;	// InlineBits.fir:2:10
  assign j = _T_0;	// InlineBits.fir:2:10
  assign k = _T_3;	// InlineBits.fir:2:10
  assign l = _T;	// InlineBits.fir:2:10
  assign m = _T_3;	// InlineBits.fir:2:10
  assign n = _T_4;	// InlineBits.fir:2:10
  assign o = _T_2;	// InlineBits.fir:2:10
  assign p = _T_1;	// InlineBits.fir:2:10
  assign q = _T_2;	// InlineBits.fir:2:10
  assign r = _T_4;	// InlineBits.fir:2:10
  assign s = _T_0;	// InlineBits.fir:2:10
  assign t = _T_2;	// InlineBits.fir:2:10
  assign u = _T_3;	// InlineBits.fir:2:10
endmodule
```

The SFC output is objectively better here:

```verilog
module InlineBits(
  input  [3:0] a,
  output       b,
  output [2:0] c,
  output [1:0] d,
  output [1:0] e,
  output [1:0] f,
  output [1:0] g,
  output [1:0] h,
  output [1:0] i,
  output [1:0] j,
  output       k,
  output       l,
  output       m,
  output       n,
  output [1:0] o,
  output [1:0] p,
  output [1:0] q,
  output       r,
  output [1:0] s,
  output [1:0] t,
  output       u
);
  assign b = a[2];
  assign c = a[3:1];
  assign d = a[3:2];
  assign e = a[1:0];
  assign f = a[3:2];
  assign g = a[2:1];
  assign h = a[1:0];
  assign i = a[2:1];
  assign j = a[3:2];
  assign k = a[3];
  assign l = a[2];
  assign m = a[3];
  assign n = a[1];
  assign o = a[2:1];
  assign p = a[1:0];
  assign q = a[2:1];
  assign r = a[1];
  assign s = a[3:2];
  assign t = a[2:1];
  assign u = a[3];
endmodule
```

lattner, 2021-07-14:
This should be a straight-forward change to PrettifyVerilog.  xref the discussion in https://github.com/llvm/circt/pull/1396

seldridge, 2022-07-03:
CIRCT is producing the exact same Verilog as the SFC on `main`. 🎊  Closing this.

closed 2022-07-03 by seldridge