llvm/circt #1417: [CI] Use the latest clang-format version

teqdruid, 2021-07-14:
Since we use a [docker-based image](https://github.com/circt/images/tree/trunk/circt-ci-build), this should be as easy as modifying the Dockerfile to install the latest clang-format version then updating the workflow in this repo to use it.

youngar, 2021-07-16:
I think [this](https://github.com/google/llvm-premerge-checks/blob/main/containers/base-debian/Dockerfile#L16) is the dockerfile for the LLVM phabricator pre-merge builds.  They install and use the latest release of `clang` as well as `clang-format`.  Would it make sense to follow suit, or do we prefer to mix the `clang` and `clang-format` versions?

teqdruid, 2021-07-20:
I don't have a preference. We should probably do what MLIR does.

youngar, 2021-08-06:
Opened a PR here: https://github.com/circt/images/pull/8

youngar, 2021-08-30:
image updated here: https://github.com/llvm/circt/commit/e16c134c035cb73f47bf311ac318ec21a8a1839a

closed 2021-08-30 by youngar