llvm/circt #1445: CIRCT doesn't compile with gcc 7.5.0

stevenmburns, 2021-07-19:
The current CIRCT main branch (a470701a49c938b8c08dfa9847234881f9595384) doesn't compile using gcc 7.5.0 (default on ubuntu 18.04).

```
(.venv) smburns@smburns-XPS-13-9370:~/circt/circt/build$ ninja -j16
[19/26] Building CXX object lib/Dialect/SV/Transforms/CMakeFiles/obj.CIRCTSVTransforms.dir/SVExtractTestCode.cpp.o
FAILED: lib/Dialect/SV/Transforms/CMakeFiles/obj.CIRCTSVTransforms.dir/SVExtractTestCode.cpp.o 
/usr/bin/c++ -DGTEST_HAS_RTTI=0 -D_DEBUG -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D__STDC_LIMIT_MACROS -I/home/smburns/circt/circt/llvm/llvm/include -I/home/smburns/circt/circt/llvm/build/include -I/home/smburns/circt/circt/llvm/mlir/include -I/home/smburns/circt/circt/llvm/build/tools/mlir/include -I/home/smburns/circt/circt/include -I/home/smburns/circt/circt/build/include -fno-exceptions -fno-rtti -fPIC -fno-semantic-interposition -fvisibility-inlines-hidden -Werror=date-time -Wall -Wextra -Wno-unused-parameter -Wwrite-strings -Wcast-qual -Wno-missing-field-initializers -Wimplicit-fallthrough -Wno-noexcept-type -Wdelete-non-virtual-dtor -Wno-comment -Wmisleading-indentation -fdiagnostics-color -ffunction-sections -fdata-sections -O3 -DNDEBUG  -fno-exceptions -fno-rtti -UNDEBUG -std=gnu++14 -MD -MT lib/Dialect/SV/Transforms/CMakeFiles/obj.CIRCTSVTransforms.dir/SVExtractTestCode.cpp.o -MF lib/Dialect/SV/Transforms/CMakeFiles/obj.CIRCTSVTransforms.dir/SVExtractTestCode.cpp.o.d -o lib/Dialect/SV/Transforms/CMakeFiles/obj.CIRCTSVTransforms.dir/SVExtractTestCode.cpp.o -c /home/smburns/circt/circt/lib/Dialect/SV/Transforms/SVExtractTestCode.cpp
/home/smburns/circt/circt/lib/Dialect/SV/Transforms/SVExtractTestCode.cpp: In function ‘void getBackwardSliceSimple(mlir::Operation*, llvm::SmallPtrSetImpl<mlir::Operation*>&, std::function<bool(mlir::Operation*)>)’:
/home/smburns/circt/circt/lib/Dialect/SV/Transforms/SVExtractTestCode.cpp:44:20: error: parse error in template argument list
     if (!op || op->hasTrait<OpTrait::IsIsolatedFromAbove>())
                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/smburns/circt/circt/lib/Dialect/SV/Transforms/SVExtractTestCode.cpp:44:59: error: no matching function for call to ‘mlir::Operation::hasTrait<<expression error> >()’
     if (!op || op->hasTrait<OpTrait::IsIsolatedFromAbove>())
                                                           ^
In file included from /home/smburns/circt/circt/llvm/mlir/include/mlir/IR/OpDefinition.h:23:0,
                 from /home/smburns/circt/circt/llvm/mlir/include/mlir/IR/FunctionSupport.h:18,
                 from /home/smburns/circt/circt/llvm/mlir/include/mlir/IR/BuiltinOps.h:16,
                 from /home/smburns/circt/circt/include/circt/Dialect/SV/SVDialect.h:18,
                 from /home/smburns/circt/circt/include/circt/Dialect/SV/SVOps.h:16,
                 from /home/smburns/circt/circt/lib/Dialect/SV/Transforms/PassDetail.h:15,
                 from /home/smburns/circt/circt/lib/Dialect/SV/Transforms/SVExtractTestCode.cpp:16:
/home/smburns/circt/circt/llvm/mlir/include/mlir/IR/Operation.h:476:53: note: candidate: template<template<class T> class Trait> bool mlir::Operation::hasTrait()
   template <template <typename T> class Trait> bool hasTrait() {
                                                     ^~~~~~~~
/home/smburns/circt/circt/llvm/mlir/include/mlir/IR/Operation.h:476:53: note:   template argument deduction/substitution failed:
/home/smburns/circt/circt/lib/Dialect/SV/Transforms/SVExtractTestCode.cpp:44:59: error: template argument 1 is invalid
     if (!op || op->hasTrait<OpTrait::IsIsolatedFromAbove>())
                                                           ^
[20/26] Linking CXX executable bin/llhd-sim
ninja: build stopped: subcommand failed.
```

This is the command:
```
cmake -G Ninja ..     -DMLIR_DIR=$PWD/../llvm/build/lib/cmake/mlir     -DLLVM_DIR=$PWD/../llvm/build/lib/cmake/llvm     -DLLVM_ENABLE_ASSERTIONS=ON     -DCMAKE_BUILD_TYPE=Release     -DCMAKE_EXPORT_COMPILE_COMMANDS=ON
ninja -j16
```

I do have success if I use the 6.0.0 version of clang.
```
CC=clang CXX=clang++ cmake -G Ninja ..     -DMLIR_DIR=$PWD/../llvm/build/lib/cmake/mlir     -DLLVM_DIR=$PWD/../llvm/build/lib/cmake/llvm     -DLLVM_ENABLE_ASSERTIONS=ON     -DCMAKE_BUILD_TYPE=Release     -DCMAKE_EXPORT_COMPILE_COMMANDS=ON
```

Should I be using clang? Or a different version of GCC?

mikeurbach, 2021-07-19:
We do support both GCC and Clang, although it looks like we are testing with a more recent version of both. The checks that run on each PR and commit to main are based on a Docker image with Ubuntu 18.04.

It uses Clang 10.0.0: https://github.com/llvm/circt/runs/3107483902#step:9:18

And GCC 9.3.0: https://github.com/llvm/circt/runs/3107483902#step:10:16

I haven't dug into this specific issue, but upgrading GCC or using Clang may be the best way to work around it. An issue was raised on the CIRCT Discord, which seems to be the same, but no resolution was mentioned. I'll cross post this issue there, in case there was any other solution.

BTW, I would recommend upgrading Clang as well, rather than using 6.0.0. I first used that version on an Ubuntu 18.04 development machine, and found compile times to be greatly reduced after upgrading to the latest Clang.

seldridge, 2021-07-22:
This should be fixed in #1469. If you still have an issue @stevenmburns, feel free to re-open.

closed 2021-07-22 by seldridge