llvm/circt #1447: [HW] TypeAlias: aggregate creation ops should support TypeAlias result types

teqdruid, 2021-07-19:
Currently, operations like `StructCreateOp` and `ArrayCreateOp` don't allow `TypeAliasType` result types. This means users have to bitcast, which isn't ideal. The `TypeAlias`es are not always used by ExportVerilog (since it is free to use the casted type or the pre-casted type).

The right way to get the correct behavior is to let aggregate creation/manipulation ops result types be type aliases. Not as important for the comb ops.

teqdruid, 2021-07-19:
@mikeurbach What exactly was the problem here?

mikeurbach, 2021-07-20:
There was nothing technicaly preventing the return type of the aggregate creation ops from being, e.g. `StructType`, rather than `StructTypeImpl`, etc. That was actually how it was implemented initially.

In review, @lattner [mentioned](https://github.com/llvm/circt/pull/1150#discussion_r662671353):

> Is there any way to get StructCreateOp to return a specific type? We shouldn't have to cast on every user.

In response, I changed all the ODS return types to the raw aggregate type (`StructTypeImpl`, rather than `StructType`). Now users don't need to cast to see through a potential type alias.

But, as you mention, some users do want to create a value of an aggregate type, and have that type be an alias. Specifically, the PyCDE work and the internal project using it, which has been the big driving factor for having this in the first place.

The bitcast solution is clever, but I personally think we should just swap back the return types in ODS. I don't have as much IR design experience, but it seems like these aggregate creation ops should be allowed to directly create an aggregate with a type alias, if a frontend like PyCDE wants to support that. Other users of these ops may have to cast to see through the potential alias, but that seems like a lesser evil than using a bitcast to introduce the alias.

Another option we briefly discussed was a dedicated op other than bitcast to introduce an alias. I'm not sure if encoding that in an extra op in the IR versus using the bitcast for it is any better.

@lattner do you have any suggestion here?

mikeurbach, 2021-07-21:
In the discussion today we all agreed with the statement in the title. I'll poke around and try to figure out a nice way to make this happen that doesn't force everyone to `type_cast` on the result of `getType`.

lattner, 2021-07-21:
Sounds good thanks.  FWIW, I think the behavior would expose itself on multi-use aggregates, e.g. if you have:

%x = struct_create : SomeType
use(x)
use (x)

Then the %x value will have to be emitted as a wire or variable in verilog, and we'd want the type "SomeType" to be used there.

mikeurbach, 2021-08-04:
I took a crack at this in https://github.com/llvm/circt/pull/1521. Besides flipping the ODS definitions back to allow these ops to create an aliased type, I resurrected my `TypeAliasOr<...>` class that I prototyped in the original PR (and abandoned). The key difference I added this time was support for implicit conversion to the template base class. This allows us to avoid some casts, for example like this one that originally started this thread: https://github.com/llvm/circt/pull/1150#discussion_r662671353. I played around with different ideas on the `TypeAliasOr<...>` class, but I'm not sure if there is a fancier C++ technique that could be applied here...

mikeurbach, 2021-09-08:
This is done via https://github.com/llvm/circt/pull/1521.

closed 2021-09-08 by mikeurbach