llvm/circt #1455: [FIRRTL] unnecessary _Temp wires

drom, 2021-07-20:
FIRTOOL produces a lot of unnecessary _T-emporal wires.

The following FIRRTL program

```python
circuit top_mod :
  module top_mod :
    input clock : Clock
    input foo : UInt<16>
    output baz0 : UInt<16>
    output baz1 : UInt<16>
    reg bar : UInt<16>, clock
    bar <= foo
    baz0 <= bar
    baz1 <= bar
```

Compiled with `firtool` produces this Verilog:

```verilog
...
module top_mod(
  input         clock,
  input  [15:0] foo,
  output [15:0] baz0, baz1);

  reg [15:0] bar;	// examples/reassign.fir:7:5
  ...

  always @(posedge clock)	// examples/reassign.fir:8:9
    bar <= foo;	// examples/reassign.fir:8:9
  wire [15:0] _T = bar;	// examples/reassign.fir:2:10
  assign baz0 = _T;	// examples/reassign.fir:2:10
  assign baz1 = _T;	// examples/reassign.fir:2:10
endmodule
```

Expected ouput:

```verilog
...
module top_mod(
  input         clock,
  input  [15:0] foo,
  output [15:0] baz0, baz1);

  reg [15:0] bar;	// examples/reassign.fir:7:5
  ...

  always @(posedge clock)	// examples/reassign.fir:8:9
    bar <= foo;	// examples/reassign.fir:8:9
  assign baz0 = bar;	// examples/reassign.fir:2:10
  assign baz1 = bar;	// examples/reassign.fir:2:10
endmodule
```

seldridge, 2021-07-20:
I agree.  It looks like the CSE is just too strong!  Similar to https://github.com/llvm/circt/issues/1416.

hanchenye, 2021-07-20:
Met similar issue. In my case, this issue is because a value is always not inlined when it has more than one use. Could you try whether this branch works? https://github.com/llvm/circt/commit/e8e275bd13304b753aa504f98f6586b36d6a481a

hanchenye, 2021-07-21:
Auh, actually this is mentioned in #1396. It seems we should duplicate `ReadInOutOp` harder in `prettify-verilog`. For now, `ReadInOutOp` is only duplicated once to the begin of a block, no matter how many times it is used in that block.