llvm/circt #1464: [SV] Make InterfaceInstanceOp more like InterfaceOp

seldridge, 2021-07-22:
SystemVerilog dialect's modeling of `InterfaceInstanceOp` is a bit of an odd duck.  _Module instantiations_ in FIRRTL and HW dialects are both function-like, but `InterfaceInstanceOp` works differently. `InterfaceInstanceOp` currently has a return type of the interface type with a symbol back to the interface declaration. 

The problem is that interfaces in SystemVerilog can be treated just like modules and instantiated.  Something like::

```verilog
MyInterface myInterface(
  .a(a),
  .b(b)
);
```

The current approach doesn't really capture this use case well.  I'd like to see some unification around "instantiation" of modules and interfaces so that things like emitting binds (which works for either interface instances or module instances) then isn't having to be handled totally differently.

See terse initial suggestion from @darthscsi here: https://github.com/llvm/circt/pull/1432#discussion_r674333321.

mikeurbach, 2021-07-22:
Could there be an op interface that both `hw.instance` and `sv.interface.instance` implement? I think the analogy from upstream MLIR's FunctionLike world would be CallOpInterface. That might not make sense to use exactly, but perhaps it could be inspiration and we could define a more hardware-centric InstantiationOpInterface, or something.

teqdruid, 2021-07-22:
Two general things to note:
- The current SV Interface support was developed with a specific use-case in mind. I personally don't care how you modify the current model so long as it continues to support the "simple communication" use-case.
- ESI makes pretty extensive use of the current SV Interface support. Please don't break it, as defined by the ESI tests and (more importantly) the ESI integration tests. If you don't have to modify the integration tests, I likely won't have a problem with any changes.