llvm/circt #1477: [FIRRTL] Add Check that Register Init Type Matches Register Type

seldridge, 2021-07-25:
Add a check that FIRRTL's `regreset` errors if the type of the reset value doesn't match the type of reset.  E.g., the following should fail because a vector register is being reset to a ground type.  However, it does not fail and will crash LowerTypes:

```mlir
%r = firrtl.regreset %clock, %reset, %c0_ui1  : (!firrtl.clock, !firrtl.asyncreset, !firrtl.uint<1>) -> !firrtl.vector<uint<1>, 1>
```

I think there's something clever that can be done here with the `AllTypesMatch` or `AllTypesMatchWith` ODS functions (or something similar). However, I'm not finding an immediate solution.

lattner, 2021-07-25:
Agreed, this should be a verifier check.

seldridge, 2022-07-03:
Fixed via 3af1bd7dc3e29127500a874013542ad5515143cc.

closed 2022-07-03 by seldridge