llvm/circt #1478: [FIRRTL] IMConstProp Bug: Output used as wire

seldridge, 2021-07-25:
One more IMConstProp bug.  Registers in instances that read from an output port are getting incorrectly optimized away. Note: this bug doesn't seem to exist for modules where the register is in the top module.  

Consider the following circuit where register `%r` reads from output port `%a` and also writes to it (effectively creating a self connection):

```mlir
firrtl.circuit "Foo"   {
  firrtl.module @Bar(in %clock: !firrtl.clock, out %a: !firrtl.uint<1>) {
    %r = firrtl.reg %clock  : (!firrtl.clock) -> !firrtl.uint<1>
    %0 = firrtl.not %a : (!firrtl.uint<1>) -> !firrtl.uint<1>
    firrtl.connect %r, %0 : !firrtl.uint<1>, !firrtl.uint<1>
    firrtl.connect %a, %r : !firrtl.uint<1>, !firrtl.uint<1>
  }
  firrtl.module @Foo(in %clock: !firrtl.clock, out %a: !firrtl.uint<1>) {
    %bar_clock, %bar_a = firrtl.instance @Bar  {name = "bar"} : !firrtl.clock, !firrtl.uint<1>
    firrtl.connect %bar_clock, %clock : !firrtl.clock, !firrtl.clock
    firrtl.connect %a, %bar_a : !firrtl.uint<1>, !firrtl.uint<1>
  }
}
```

IMConstProp is deleting the register `%r` when it should keep it around.  After IMConstProp, the circuit is:

```mlir
firrtl.circuit "Foo"   {
  firrtl.module @Bar(in %clock: !firrtl.clock, out %a: !firrtl.uint<1>) {
    %invalid_ui1 = firrtl.invalidvalue : !firrtl.uint<1>
    firrtl.connect %a, %invalid_ui1 : !firrtl.uint<1>, !firrtl.uint<1>
  }
  firrtl.module @Foo(in %clock: !firrtl.clock, out %a: !firrtl.uint<1>) {
    %invalid_ui1 = firrtl.invalidvalue : !firrtl.uint<1>
    %bar_clock, %bar_a = firrtl.instance @Bar  {name = "bar"} : !firrtl.clock, !firrtl.uint<1>
    firrtl.connect %bar_clock, %clock : !firrtl.clock, !firrtl.clock
    firrtl.connect %a, %invalid_ui1 : !firrtl.uint<1>, !firrtl.uint<1>
  }
}
```

### Background

The source FIRRTL for this is:

```scala
circuit Foo :
  module Bar :
    input clock : Clock
    output a : UInt<1>

    reg r : UInt<1>, clock
    r <= not(a)
    a <= r

  module Foo :
    input clock : Clock
    output a : UInt<1>

    inst bar of Bar
    bar.clock <= clock
    a <= bar.a
```

Running this through the Scala FIRRTL Compiler produces the following:

```verilog
module Bar(
  input   clock,
  output  a
);
  reg  r;
  assign a = r;
  always @(posedge clock) begin
    r <= ~a;
  end
endmodule
module Foo(
  input   clock,
  output  a
);
  wire  bar_clock;
  wire  bar_a;
  Bar bar (
    .clock(bar_clock),
    .a(bar_a)
  );
  assign a = bar_a;
  assign bar_clock = clock;
endmodule

```

Running this through CIRCT produces:

```verilog
module Bar(
  input  clock,
  output a);

  assign a = 1'h0;	// Foo.fir:2:10, :6:5
endmodule

module Foo(
  input  clock,
  output a);

  wire bar_a;	// Foo.fir:14:5

  Bar bar (	// Foo.fir:14:5
    .clock (clock),
    .a     (bar_a)
  );
  assign a = 1'h0;	// Foo.fir:10:10, :14:5
endmodule
```

I was surprised that formulations where the register is in the top module work. I.e., the following compiles just fine:

```scala
circuit Foo :
  module Foo :
    input clock : Clock
    output a : UInt<1>

    reg r : UInt<1>, clock
    r <= not(a)
    a <= r
```

seldridge, 2021-07-26:
Thinking about this a bit more, this may be highlighting some weirdness in the SFC, too. 🤔 Generally, the SFC will always optimize away a register that is either unconnected or driven only by itself.  However, it deviates from this if the path forming the self connection involves an output port because SFC constant prop doesn't appear to be recursing to look at output port connections.

Consider the following formulations, all involving register `r`. Here, the SFC _will not optimize away r_.:

```scala
output a: UInt<1>
r <= a
a <= r
```  

Here, the SFC _will optimize away r (noting that `wire a` will become a node during SFC lowering):

```scala
wire a: UInt<1>
r <= a
a <= r
```

The case of `not(a)` shouldn't be optimized away, but this is for the different reason that this involves a primop which can't be looked across.

seldridge, 2021-07-26:
Also, this appears to be related to #1465, but not occurring in the top module.  SFC oddities are being tracked in https://github.com/chipsalliance/firrtl/issues/2299.

lattner, 2021-07-27:
We discussed this today, and agreed we should start regs/mems as being "overdefined" in their initial state.

seldridge, 2021-07-28:
Sorry, this took more time to get together... #1486 overdefines all `RegOp` and `RegResetOp`.  (All the examples above only show `RegOp`, but the problem is also there for `RegResetOp`.)  The approach in #1481 is a non-starter due to the defined folder on `xor(x, invalid) -> invalid` which prunes oscillators that use the `xor(a, UInt<1>)` formulation.

commit c5c6f082 (2021-07-27): [FIRRTL] Invalidate Module Outputs in IMCP

Invalidate all ground-type module outputs (which are not the top module)
for circuits where the top module is a module.  This is done to treat
these ports (which are sinks/lvalues) like wires when they are used as
RHS sources/rvalues.  All aggregate module outputs are marked
overdefined.

This blocks removal of the register "r" in the following FIRRTL
module (when this module is NOT the top module).  Without this patch,
output "a" never receives a lattice value:

    module Foo:
      input clock: Clock
      output a: UInt<1>

      reg r: UInt<1>,  [... cut]
commit d9c1af0b (2021-07-28): [FIRRTL] Invalidate Module Outputs in IMCP

Invalidate all ground-type module outputs (which are not the top module)
for circuits where the top module is a module.  This is done to treat
these ports (which are sinks/lvalues) like wires when they are used as
RHS sources/rvalues.  All aggregate module outputs are marked
overdefined.

This blocks removal of the register "r" in the following FIRRTL
module (when this module is NOT the top module).  Without this patch,
output "a" never receives a lattice value:

    module Foo:
      input clock: Clock
      output a: UInt<1>

      reg r: UInt<1>,  [... cut]
PR #1490 (2021-07-28): [FIRRTL] Update output ports in IMCP lattice
Change IMCP to update output ports and not just _external_ users of
output ports.  Output ports have wire-like semantics and may be read
from as well as written to.

Fixes #1478.
Fixes #1488.
closed 2021-07-28 by seldridge