llvm/circt #1483: [FIRRTL][ConstantProp] Constant propagation and expansion of muxes

prithayan, 2021-07-27:
The following case of constant propagation of registers and speculative expansion of optimized muxes is not handled by circt.
The following FIRRTL program
```scala
circuit Top :
  module Top :
    input clock : Clock
    input en : UInt<1>
    output out : UInt<1>
    reg r1 : UInt<1>, clock
    reg r2 : UInt<1>, clock
    when en :
      r1 <= UInt<1>(1)
    r2 <= UInt<1>(0)
    when en :
      r2 <= r2
    out <= xor(r1, r2)
```
`firrtl-1.5-SNAPSHOT` produces the following verilog
```verilog
module Top(
  input   clock,
  input   en,
  output  out
);
  assign out = 1'h1;
endmodule
```
`firtool` produces the following verilog
```verilog
module Top(
  input  clock, en,
  output out);
  reg r2; // 26.fir:7:5
  `ifndef SYNTHESIS // 26.fir:7:5
    initial begin // 26.fir:7:5
      automatic logic [31:0] _T_1;  // 26.fir:7:5
      `INIT_RANDOM_PROLOG_  // 26.fir:7:5
      `ifdef RANDOMIZE_REG_INIT // 26.fir:7:5
        _T_1 = `RANDOM; // 26.fir:7:5
        r2 = _T_1[0]; // 26.fir:7:5
      `endif
    end // initial
  `endif
  wire _T = r2; // 26.fir:12:10
  wire _T_0 = en & _T;  // 26.fir:12:10
  always @(posedge clock) // 26.fir:12:10
    r2 <= _T_0; // 26.fir:12:10
  assign out = ~_T; // 26.fir:2:10, :13:12
endmodule
```

prithayan, 2021-07-27:
Smaller test case, 
`firtool` is not able to constant prop this case too,
``` scala
circuit Top :
  module Top :
    input clock : Clock
    input reset : UInt<1>
    input cond : UInt<1>
    output z : UInt<8>
    reg r : UInt<8>, clock with : (reset => (reset, UInt<4>("hb")))
    when cond :
      r <= UInt<4>("hb")
    z <= r
```
`firttl` produces
```
module Top(
  input        clock,
  input        reset,
  input        cond,
  output [7:0] z
);
  assign z = 8'hb;
endmodule
```

fyquah, 2021-07-28:
> ```scala
> circuit Top :
>   module Top :
>     input clock : Clock
>     input en : UInt<1>
>     output out : UInt<1>
>     reg r1 : UInt<1>, clock
>     reg r2 : UInt<1>, clock
>     when en :
>       r1 <= UInt<1>(1)
>     r2 <= UInt<1>(0)
>     when en :
>       r2 <= r2
>     out <= xor(r1, r2)
> ```

Assuming uninitialized registers are initialized to zero, is the correct behavior here `t=0, r2=0`, `t=1, r2=1`?

seldridge, 2021-07-28:
> Assuming uninitialized registers are initialized to zero, is the correct behavior here `t=0, r2=0`, `t=1, r2=1`?

They're actually uninitialized and not initialized to zero. This is what allows the compiler to optimize it. If it's initialized to zero, you can only remove `r2` and not `r1`. 

Right before SFC constant prop the circuit looks like the following.  Each register can only be uninitialized or a constant. Note that the `reset => (UInt<1>("h0"), r1)` is really a verbose way of saying "this register has no reset because its reset line is tied to zero and its reset initial value is itself":

```scala
    reg r1 : UInt<1>, clock with :
      reset => (UInt<1>("h0"), r1)
    reg r2 : UInt<1>, clock with :
      reset => (UInt<1>("h0"), r2)
    node _GEN_0 = mux(en, UInt<1>("h1"), r1)
    node _GEN_1 = mux(en, r2, UInt<1>("h0"))
    out <= xor(r1, r2)
    r1 <= _GEN_0
    r2 <= _GEN_1
```

Each register only has two possible drivers: (1) itself/uninit and (2) constant.  These can be combined to just the constant:

```scala
    reg r1 : UInt<1>, clock with :
      reset => (UInt<1>("h0"), r1)
    reg r2 : UInt<1>, clock with :
      reset => (UInt<1>("h0"), r2)
    node _GEN_0 = UInt<1>("h1")
    node _GEN_1 = UInt<1>("h0")
    out <= UInt<1>("h1")
    r1 <= UInt<1>("h1")
    r2 <= UInt<1>("h0")
```

Further constant propping into `out` then makes both `r1` and `r2` dead code and they get dropped.

If there's an assumption about `t=0` being zero, then this optimization isn't possible. In that case, you can only remove `r2` (due to it's other driver being zero).  If you add a reset line to the registers to model this, the final circuit will be:

```scala
    reg r1 : UInt<1>, clock with :
      reset => (reset, UInt<1>("h0"))
    out <= r1
    r1 <= or(en, r1)
```

commit bd7edb5c (2021-08-10): [FIRRTL] Constant prop registers with self-init to mux pattern (#1536)

Constant propagate registers that are driven by a mux tree 
containing only instances of one constant and self-assigns.

Detect the case if a register only has two possible drivers: 
(1) itself/uninit and (2) constant.
The mux can then be replaced with the constant.

reg = mux(cond1, reg, 3) can be replaced with --> reg = 3
reg = mux(cond2, 3, reg) can be replaced with --> reg = 3

Fixes #1483
closed 2021-08-10 by prithayan