llvm/circt #1509: [PyCDE] Create pipeline to create and publish python wheel

teqdruid, 2021-08-02:
Consider using https://cibuildwheel.readthedocs.io/en/stable/

Use npcomp as an example: https://github.com/llvm/mlir-npcomp/pull/256/files

teqdruid, 2022-08-03:
We did this internally so as to use faster build agents and pypi auth issues.

hovind, 2024-04-13:
@teqdruid The `pycde` wheel hasn't been updated since Sep 22, 2023. Could we re-open this issue?

I imagine that the `pypi` auth issues have been sorted out given that wheels for the `esi-runtime` and python `circt` bindings are being published.

teqdruid, 2024-04-15:
Actually, it's failing since the pycde project is out of space on pypi and I haven't had the chance to delete old daily wheels.

But now that the pypi thing has been figures out and we can run the builds out of the circt repo on the fast runners, building and publishing out of circt is an option. In fact, it's something I've been meaning to get around to.

teqdruid, 2024-04-15:
And it's only been failing for the last two weeks or so. I wouldn't be surprised if the last versioned release was back in September though. Look through the pre releases for the nightlies.

hovind, 2024-04-15:
> And it's only been failing for the last two weeks or so. I wouldn't be surprised if the last versioned release was back in September though. Look through the pre releases for the nightlies.

Ah, there it is, thank you! I'm new to the python ecosystem, apologies for the confusion. Thanks for the quick clarification.

PR #7062 (2024-05-21): [PyCDE] Move build and publishing pipeline into repo
Previously was built and published internally. Resolves #1509.
commit 2ba19927 (2024-05-21): [PyCDE] Move build and publishing pipeline into repo (#7062)

Previously was built and published internally. Resolves #1509.
closed 2024-05-21 by teqdruid