llvm/circt #1511: [FIRRTL] (code dump) type/value width mismatch materializing constant

drom, 2021-08-03:
The following FIRRTL program:

```python
; seed: 133
circuit top_mod :
  module mod_1 :
    input clock: Clock
    input reset: Reset
    input arst: AsyncReset
    input inp_bb: SInt<26>[1][4]
    reg tmp61: UInt<26>, clock with: (reset => (arst, UInt<26>(0)))
    tmp61 <= xor(SInt(0), inp_bb[3][0])
  module top_mod :
    input clock: Clock
    input reset: UInt<1>
    input arst: AsyncReset
    inst U1 of mod_1
    U1 is invalid
    inst U2 of mod_1
    U2 is invalid
    ; U1
    U1.reset <= reset
    U1.arst <= arst
    U1.clock <= clock
    ; U2
    U2.reset <= reset
    U2.arst <= arst
    U2.clock <= clock
```

Compiled with `valgrind -q $FIRTOOL $INPUT --lower-to-hw --infer-widths --mlir-timing --verilog -o=$VFILE2` fails with the following core dump:

```
firtool: /home/drom/work/github/llvm/circt/lib/Dialect/FIRRTL/FIRRTLDialect.cpp:325: virtual mlir::Operation* circt::firrtl::FIRRTLDialect::materializeConstant(mlir::OpBuilder&, mlir::Attribute, mlir::Type, mlir::Location): Assertion `(!intType.hasWidth() || (unsigned)intType.getWidthOrSentinel() == attrValue.getValue().getBitWidth()) && "type/value width mismatch materializing constant"' failed.
PLEASE submit a bug report to https://bugs.llvm.org/ and include the crash backtrace.
Stack dump:
0.      Program arguments: ../../llvm/circt/build/bin/firtool a_top_mod.fir --lower-to-hw --infer-widths --mlir-timing --verilog -o=a_top_mod.fir.mfc.v
1.       #0 0x00000000004e2be7 llvm::sys::PrintStackTrace(llvm::raw_ostream&, int) (.localalias) /home/drom/work/github/llvm/circt/llvm/llvm/lib/Support/Unix/Signals.inc:565:22
 #1 0x00000000004e2caa PrintStackTraceSignalHandler(void*) /home/drom/work/github/llvm/circt/llvm/llvm/lib/Support/Unix/Signals.inc:632:1
 #2 0x00000000004e0bf1 llvm::sys::RunSignalHandlers() (.localalias) /home/drom/work/github/llvm/circt/llvm/llvm/lib/Support/Signals.cpp:97:20
 #3 0x00000000004e2612 SignalHandler(int) /home/drom/work/github/llvm/circt/llvm/llvm/lib/Support/Unix/Signals.inc:407:1
 #4 0x000000000488b970 __restore_rt (/lib64/libpthread.so.0+0x13970)
 #5 0x0000000004a93322 raise (/lib64/libc.so.6+0x3d322)
 #6 0x0000000004a7c864 abort (/lib64/libc.so.6+0x26864)
 #7 0x0000000004a7c749 _nl_load_domain.cold (/lib64/libc.so.6+0x26749)
 #8 0x0000000004a8b9d6 (/lib64/libc.so.6+0x359d6)
 #9 0x0000000000895621 circt::firrtl::FIRRTLDialect::materializeConstant(mlir::OpBuilder&, mlir::Attribute, mlir::Type, mlir::Location) (.localalias) /home/drom/work/github/llvm/circt/lib/Dialect/FIRRTL/FIRRTLDialect.cpp:325:5
#10 0x000000000064c9c6 (anonymous namespace)::IMConstPropPass::rewriteModuleBody(circt::firrtl::FModuleOp)::'lambda'(mlir::Value)::operator()(mlir::Value) const /home/drom/work/github/llvm/circt/lib/Dialect/FIRRTL/Transforms/IMConstProp.cpp:661:58
#11 0x000000000064cdeb (anonymous namespace)::IMConstPropPass::rewriteModuleBody(circt::firrtl::FModuleOp) /home/drom/work/github/llvm/circt/lib/Dialect/FIRRTL/Transforms/IMConstProp.cpp:716:17
#12 0x000000000064e522 auto (anonymous namespace)::IMConstPropPass::runOnOperation()::'lambda'(auto)::operator()<circt::firrtl::FModuleOp>(auto) const /home/drom/work/github/llvm/circt/lib/Dialect/FIRRTL/Transforms/IMConstProp.cpp:313:63
#13 0x000000000064e555 auto void mlir::parallelForEach<mlir::detail::op_iterator<circt::firrtl::FModuleOp, llvm::ilist_iterator<llvm::ilist_detail::node_options<mlir::Operation, true, false, void>, false, false> >, (anonymous namespace)::IMConstPropPass::runOnOperation()::'lambda'(auto)>(mlir::MLIRContext*, auto, auto, (anonymous namespace)::IMConstPropPass::runOnOperation()::'lambda'(auto)&&)::'lambda'(auto&&)::operator()<circt::firrtl::FModuleOp>(auto&&) const /home/drom/work/github/llvm/circt/llvm/mlir/include/mlir/IR/Threading.h:125:63
#14 0x000000000064e609 mlir::LogicalResult mlir::failableParallelForEach<mlir::detail::op_iterator<circt::firrtl::FModuleOp, llvm::ilist_iterator<llvm::ilist_detail::node_options<mlir::Operation, true, false, void>, false, false> >, void mlir::parallelForEach<mlir::detail::op_iterator<circt::firrtl::FModuleOp, llvm::ilist_iterator<llvm::ilist_detail::node_options<mlir::Operation, true, false, void>, false, false> >, (anonymous namespace)::IMConstPropPass::runOnOperation()::'lambda'(auto)>(mlir::MLIRContext*, auto, auto, (anonymous namespace)::IMConstPropPass::runOnOperation()::'lambda'(auto)&&)::'lambda'(auto&&)>(mlir::MLIRContext*, auto, auto, (anonymous namespace)::IMConstPropPass::runOnOperation()::'lambda'(auto)&&)::'lambda'()::operator()() const /home/drom/work/github/llvm/circt/llvm/mlir/include/mlir/IR/Threading.h:65:17
#15 0x000000000064e8a3 mlir::LogicalResult mlir::failableParallelForEach<mlir::detail::op_iterator<circt::firrtl::FModuleOp, llvm::ilist_iterator<llvm::ilist_detail::node_options<mlir::Operation, true, false, void>, false, false> >, void mlir::parallelForEach<mlir::detail::op_iterator<circt::firrtl::FModuleOp, llvm::ilist_iterator<llvm::ilist_detail::node_options<mlir::Operation, true, false, void>, false, false> >, (anonymous namespace)::IMConstPropPass::runOnOperation()::'lambda'(auto)>(mlir::MLIRContext*, auto, auto, (anonymous namespace)::IMConstPropPass::runOnOperation()::'lambda'(auto)&&)::'lambda'(auto&&)>(mlir::MLIRContext*, auto, auto, (anonymous namespace)::IMConstPropPass::runOnOperation()::'lambda'(auto)&&) /home/drom/work/github/llvm/circt/llvm/mlir/include/mlir/IR/Threading.h:81:3
#16 0x000000000064de44 void mlir::parallelForEach<mlir::detail::op_iterator<circt::firrtl::FModuleOp, llvm::ilist_iterator<llvm::ilist_detail::node_options<mlir::Operation, true, false, void>, false, false> >, (anonymous namespace)::IMConstPropPass::runOnOperation()::'lambda'(auto)>(mlir::MLIRContext*, auto, auto, (anonymous namespace)::IMConstPropPass::runOnOperation()::'lambda'(auto)&&) /home/drom/work/github/llvm/circt/llvm/mlir/include/mlir/IR/Threading.h:124:32
#17 0x000000000064d4c0 void mlir::parallelForEach<llvm::iterator_range<mlir::detail::op_iterator<circt::firrtl::FModuleOp, llvm::ilist_iterator<llvm::ilist_detail::node_options<mlir::Operation, true, false, void>, false, false> > >, (anonymous namespace)::IMConstPropPass::runOnOperation()::'lambda'(auto)>(mlir::MLIRContext*, auto&&, (anonymous namespace)::IMConstPropPass::runOnOperation()::'lambda'(auto)&&) /home/drom/work/github/llvm/circt/llvm/mlir/include/mlir/IR/Threading.h:136:18
#18 0x000000000064b13c (anonymous namespace)::IMConstPropPass::runOnOperation() /home/drom/work/github/llvm/circt/lib/Dialect/FIRRTL/Transforms/IMConstProp.cpp:316:17
#19 0x0000000000ba989d mlir::detail::OpToOpPassAdaptor::run(mlir::Pass*, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int) (.localalias) /home/drom/work/github/llvm/circt/llvm/mlir/lib/Pass/Pass.cpp:386:27
#20 0x0000000000ba9b13 mlir::detail::OpToOpPassAdaptor::runPipeline(llvm::iterator_range<llvm::pointee_iterator<std::unique_ptr<mlir::Pass, std::default_delete<mlir::Pass> >*, mlir::Pass> >, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int, mlir::PassInstrumentor*, mlir::PassInstrumentation::PipelineParentInfo const*) (.localalias) /home/drom/work/github/llvm/circt/llvm/mlir/lib/Pass/Pass.cpp:445:15
#21 0x0000000000bac665 auto mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::'lambda'(auto&)::operator()<std::pair<mlir::Operation*, mlir::AnalysisManager> >(auto&) const /home/drom/work/github/llvm/circt/llvm/mlir/lib/Pass/Pass.cpp:607:20
#22 0x0000000000bac84e mlir::LogicalResult mlir::failableParallelForEach<__gnu_cxx::__normal_iterator<std::pair<mlir::Operation*, mlir::AnalysisManager>*, std::vector<std::pair<mlir::Operation*, mlir::AnalysisManager>, std::allocator<std::pair<mlir::Operation*, mlir::AnalysisManager> > > >, mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::'lambda'(auto&)&>(mlir::MLIRContext*, auto, auto, mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::'lambda'(auto&)&) /home/drom/work/github/llvm/circt/llvm/mlir/include/mlir/IR/Threading.h:49:17
#23 0x0000000000bac021 mlir::LogicalResult mlir::failableParallelForEach<std::vector<std::pair<mlir::Operation*, mlir::AnalysisManager>, std::allocator<std::pair<mlir::Operation*, mlir::AnalysisManager> > >&, mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::'lambda'(auto&)&>(mlir::MLIRContext*, auto&&, mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::'lambda'(auto&)&) /home/drom/work/github/llvm/circt/llvm/mlir/include/mlir/IR/Threading.h:99:1
[... cut ...]
#31 0x0000000000406c47 executeFirtool(mlir::MLIRContext&) /home/drom/work/github/llvm/circt/tools/firtool/firtool.cpp:377:30
#32 0x0000000000406e29 main /home/drom/work/github/llvm/circt/tools/firtool/firtool.cpp:413:31
#33 0x0000000004a7db35 __libc_start_main (/lib64/libc.so.6+0x27b35)
#34 0x000000000040584e _start /home/abuild/rpmbuild/BUILD/glibc-2.33/csu/../sysdeps/x86_64/start.S:122:0
./process1.sh: line 43: 16799 Aborted                 (core dumped) valgrind -q $FIRTOOL $INPUT --lower-to-hw --infer-widths --mlir-timing --verilog -o=$VFILE2
f```

seldridge, 2021-08-03:
Awesome! Nice find @drom. 

Here is a more reduced version:

```scala
circuit top_mod :
  module top_mod :
    output b: UInt<2>
    wire a : SInt<2>
    a is invalid
    b <= xor(SInt<1>(0), a)
```

And the problematic MLIR:
```scala
firrtl.circuit "top_mod"   {
  firrtl.module @top_mod(out %b: !firrtl.uint<2>) {
    %c0_si1 = firrtl.constant 0 : !firrtl.sint<1>
    %invalid_si2 = firrtl.invalidvalue : !firrtl.sint<2>
    %0 = firrtl.xor %c0_si1, %invalid_si2 : (!firrtl.sint<1>, !firrtl.sint<2>) -> !firrtl.uint<2>
    firrtl.connect %b, %0 : !firrtl.uint<2>, !firrtl.uint<2>
  }
}
```

This appears to be both having problems with the 1-bit SInt and trying to create a signed constant where it needs to create an unsigned one.  There is a similar failure, but without the crash if the widths are the same:

```scala
circuit top_mod :
  module top_mod :
    output b: UInt<2>
    wire a : SInt<2>
    a is invalid
    b <= xor(SInt<2>(0), a)
```

This will fail with:

```
1511.fir:6:10: error: firrtl.constant attribute has wrong sign
    b <= xor(SInt<2>(0), a)
         ^
1511.fir:6:10: note: see current operation: %0 = "firrtl.constant"() {value = 0 : si2} : () -> !firrtl.uint<2>
```

seldridge, 2021-08-03:
I think I know what's going on here.  The folds that were added (_that I added_) in https://github.com/llvm/circt/pull/1499 are returning the non-invalid operand.  This won't work if the type of the invalid operand doesn't match the result type.  I'll add checks to block this fold so this will stop failing.  After that, we define new canonicalizers that can do the casting here.

seldridge, 2021-08-03:
Fixed in ee0f736e52f01e964d4f55de66b9197a244155a2.  This is the naive solution of blocking folding if the ops don't match.

closed 2021-08-03 by seldridge